toy/blueutil #4: Conflicting usage information

PaulCapestany, 2017-01-22:
Just wanted to give you a heads up that doing `brew info blueutil` show a link to http://www.frederikseiffert.de/blueutil/ which happens to have pretty outdated usage information (from a superficial scan of your commits it seems like it'd probably be around cc07347fc0547aba5b29b98f4cc0753608fab375). 

Not a big deal, though in my case it briefly made me think this utility was broken—luckily it's not though, thanks for making it! 👍

toy, 2017-01-22:
You've triggered a [cleanup](https://github.com/toy/blueutil/compare/cleanup?expand=1) :D
Would be awesome if you could check if `make` runs successfully for you on branch `cleanup` and especially if you could check that brew works with this changes.

PaulCapestany, 2017-01-23:
Nice! RE: running `make` on the cleanup branch, it seemed like `-Werror` was preventing it from working for me, see https://github.com/toy/blueutil/pull/5

RE: homebrew, it seems that formulas have changed a bit since the last time I submitted one, will see if I read through https://github.com/Homebrew/brew/blob/master/docs/Formula-Cookbook.md to refresh my memory.

toy, 2017-01-23:
Fun to debug such stuff using a bluetooth keyboard/touchpad :D

toy, 2017-01-23:
But the warning were important ones, unfortunately not reported by my gcc. Too much ruby for me, I've forgot how `||` works in C.

commit 9df828cc (2017-01-23): add original style arguments: status, on, off, resolves #4
commit 3be72f85 (2017-02-01): add original style arguments: status, on, off, resolves #4
commit 23a59ddc (2017-02-01): add original style arguments: status, on, off, resolves #4
closed 2017-02-01 by toy