toy/blueutil #6: The new makefile doesn't work

chdiza, 2017-02-02:
On Sierra 10.12.3, if I type `make` while in the unzipped source directory, I get this:
```
cc -Wall -Wextra -Werror -framework IOBluetooth    blueutil.m   -o blueutil
blueutil.m:86:8: error: size argument in 'strncmp' call is a comparison [-Werror,-Wmemsize-comparison]
                        if (is_abbr_arg("help", argv[1])) {
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
blueutil.m:73:68: note: expanded from macro 'is_abbr_arg'
#define is_abbr_arg(name, arg) (strncmp((name), (arg), strlen(arg) || 1) == 0)
                                                       ~~~~~~~~~~~~^~~~
blueutil.m:86:8: note: did you mean to compare the result of 'strncmp' instead?
blueutil.m:73:33: note: expanded from macro 'is_abbr_arg'
#define is_abbr_arg(name, arg) (strncmp((name), (arg), strlen(arg) || 1) == 0)
                                ^
blueutil.m:86:8: note: explicitly cast the argument to size_t to silence this warning
blueutil.m:73:56: note: expanded from macro 'is_abbr_arg'
#define is_abbr_arg(name, arg) (strncmp((name), (arg), strlen(arg) || 1) == 0)
                                                       ^
blueutil.m:90:8: error: size argument in 'strncmp' call is a comparison [-Werror,-Wmemsize-comparison]
                        if (is_abbr_arg("version", argv[1])) {
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
blueutil.m:73:68: note: expanded from macro 'is_abbr_arg'
#define is_abbr_arg(name, arg) (strncmp((name), (arg), strlen(arg) || 1) == 0)
                                                       ~~~~~~~~~~~~^~~~
blueutil.m:90:8: note: did you mean to compare the result of 'strncmp' instead?
blueutil.m:73:33: note: expanded from macro 'is_abbr_arg'
#define is_abbr_arg(name, arg) (strncmp((name), (arg), strlen(arg) || 1) == 0)
                                ^
blueutil.m:90:8: note: explicitly cast the argument to size_t to silence this warning
blueutil.m:73:56: note: expanded from macro 'is_abbr_arg'
#define is_abbr_arg(name, arg) (strncmp((name), (arg), strlen(arg) || 1) == 0)
                                                       ^
blueutil.m:94:8: error: size argument in 'strncmp' call is a comparison [-Werror,-Wmemsize-comparison]
                        if (is_abbr_arg("status", argv[1])) {
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
blueutil.m:73:68: note: expanded from macro 'is_abbr_arg'
#define is_abbr_arg(name, arg) (strncmp((name), (arg), strlen(arg) || 1) == 0)
                                                       ~~~~~~~~~~~~^~~~
blueutil.m:94:8: note: did you mean to compare the result of 'strncmp' instead?
blueutil.m:73:33: note: expanded from macro 'is_abbr_arg'
#define is_abbr_arg(name, arg) (strncmp((name), (arg), strlen(arg) || 1) == 0)
                                ^
blueutil.m:94:8: note: explicitly cast the argument to size_t to silence this warning
blueutil.m:73:56: note: expanded from macro 'is_abbr_arg'
#define is_abbr_arg(name, arg) (strncmp((name), (arg), strlen(arg) || 1) == 0)
                                                       ^
blueutil.m:109:8: error: size argument in 'strncmp' call is a comparison [-Werror,-Wmemsize-comparison]
                        if (is_abbr_arg("power", argv[1])) {
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
blueutil.m:73:68: note: expanded from macro 'is_abbr_arg'
#define is_abbr_arg(name, arg) (strncmp((name), (arg), strlen(arg) || 1) == 0)
                                                       ~~~~~~~~~~~~^~~~
blueutil.m:109:8: note: did you mean to compare the result of 'strncmp' instead?
blueutil.m:73:33: note: expanded from macro 'is_abbr_arg'
#define is_abbr_arg(name, arg) (strncmp((name), (arg), strlen(arg) || 1) == 0)
                                ^
blueutil.m:109:8: note: explicitly cast the argument to size_t to silence this warning
blueutil.m:73:56: note: expanded from macro 'is_abbr_arg'
#define is_abbr_arg(name, arg) (strncmp((name), (arg), strlen(arg) || 1) == 0)
                                                       ^
blueutil.m:112:15: error: size argument in 'strncmp' call is a comparison [-Werror,-Wmemsize-comparison]
                        } else if (is_abbr_arg("discoverable", argv[1])) {
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
blueutil.m:73:68: note: expanded from macro 'is_abbr_arg'
#define is_abbr_arg(name, arg) (strncmp((name), (arg), strlen(arg) || 1) == 0)
                                                       ~~~~~~~~~~~~^~~~
blueutil.m:112:15: note: did you mean to compare the result of 'strncmp' instead?
blueutil.m:73:33: note: expanded from macro 'is_abbr_arg'
#define is_abbr_arg(name, arg) (strncmp((name), (arg), strlen(arg) || 1) == 0)
                                ^
blueutil.m:112:15: note: explicitly cast the argument to size_t to silence this warning
blueutil.m:73:56: note: expanded from macro 'is_abbr_arg'
#define is_abbr_arg(name, arg) (strncmp((name), (arg), strlen(arg) || 1) == 0)
                                                       ^
5 errors generated.
make: *** [blueutil] Error 1
```

I have Xcode 8.2.1 and the latest CommandLineTools installed.

chdiza, 2017-02-02:
I should add that, if I do it the `xcodebuild` way, I see similar messages, but they are only dubbed "warnings" and the build allegedly finishes

toy, 2017-02-02:
Thank you, please confirm the fix

chdiza, 2017-02-02:
It built, but then to my surprise it auto-launched some kind of test suite that turned off my bluetooth and thus left me with neither keyboard or mouse.  Maybe that should be behind a `make install` target.  I was not expecting that, since `xcodebuild` doesn't do that to me.

Thanks for the fix.

toy, 2017-02-03:
Sorry for the mess, I hope bluetooth was successfully restored to original state.
I've updated the Makefile to only build by default and added other targets to test and install.

chdiza, 2017-02-03:
It was restored, fortunately.  I can imagine something going wrong during the test, in such a way that bluetooth is never turned back on.  Maybe a warning should be printed to the user on stdout if they ask for `make test`.

(Maybe there is an easy way to turn bluetooth back on if one's keyboard and mouse have been disconnected, but short of having a wired mouse, or perhaps rebooting and hoping the OS attempts to find one's gear, I don't know what that is.)

chdiza, 2017-02-03:
BTW, `blueutil` is a nice little tool.  I like it, and I'm glad it's still maintained.

closed 2017-02-03 by chdiza

toy, 2017-02-03:
Good idea, quick search suggests that turning off bluetooth without wired input devices at hand will not be a nice experience, so I've added a warning + confirmation: fd6999bd4334b8c0ba58e3762500625d63e5d120.
I am really glad that the tool is helpful :)

commit 68d75e6c (2017-02-02): fix wrong handling of length in is_abbr_arg, fixes #6
commit d74d9670 (2017-02-04): fix wrong handling of length in is_abbr_arg, fixes #6