toy/blueutil #7: Compilation and proposals

ekuester, 2018-01-29:
I just compiled the code with Xcode 8.3.3 under Sierra without problems. It is mentioned that the project file is for Xcode 3.1 and that is recommended to update to newer settings. I did so and still there were not any problems. I copied the resulting binary "blueutil" into /usr/local/bin, which is in my PATH and have the command at my finger tips ... One proposal from my side: to prefix the options with a "-" which is normally the standard in Unix environments. Would look like
`blueutil -s`

toy, 2018-01-29:
I'm glad that it is useful :D

One option is to just allow prefixing with `-`.

Another is to cleanup possible arguments and bump version to 2.0.
From:
```
blueutil h[elp] - this help
blueutil v[ersion] - show version

blueutil - show state
blueutil p[ower]|d[iscoverable] - show state 1 or 0
blueutil p[ower]|d[iscoverable] 1|0 - set state

Also original style arguments:
blueutil s[tatus] - show status
blueutil on - power on
blueutil off - power off
```
to something like:
```
blueutil -h|--help - this help
blueutil -v|--version - show version

blueutil - show state
blueutil -p|--power - power state (1 or 0)
blueutil -d|--discoverable - discoverability (1 or 0)
blueutil -p|--power 1|0 - set power state
blueutil -d|--discoverable 1|0 - set discoverability
```
I don't include `-s` or `--status` as it was the original argument which had to be split into power and discoverability.

ekuester, 2018-01-30:
I would prefer a new release 2.0 :-) But it is clearly your decision, of course. Eventually instead 0 | 1 also on | off could be allowed (in my eyes programmatically no great effort). Thanks for sharing the program.

toy, 2018-02-17:
@ekuester Can you try the [v2 branch](https://github.com/toy/blueutil/compare/v2)? Does the interface feel right to you?

ekuester, 2018-02-17:
At first glance ... looks for me just fine ... all seems to work.

toy, 2018-02-18:
Merged to master and tagged v2.0.0, thanks for opening the issue

commit a6c2850d (2018-02-18): change arguments specification, resolve #7
commit 5cbc75be (2018-02-18): change arguments specification, resolve #7
commit b4c2231b (2018-02-18): update xcode project, resolve #7 (missing part)