toy/blueutil #8: Add a "--power toggle" option

lhaeger, 2018-04-18:
In order to utilize blueutil with keyboard tools like Better Touch Tools or similar, a command line option to toggle the power state would be helpful. Basically the same as an applescript like this, just without the need for a script:

set blueutil to "/usr/local/bin/blueutil"
set powerStatus to do shell script blueutil & " --power"

if powerStatus is "1" then
	do shell script blueutil & " --power off"
else if powerStatus is "0" then
	do shell script blueutil & " --power on"
end if

toy, 2018-04-18:
Please have a look/check how d4b456b9006d16cf72cc472f3afee19e1b8f0777 works for you.

lhaeger, 2018-04-19:
Seems to do the trick rather nicely. Thanks a lot, Ivan!

closed 2018-04-19 by lhaeger

toy, 2018-04-19:
v2.1.0 is out

commit d4b456b9 (2018-04-19): add possibility to toggle power and discoverability, resolves #8