toy/blueutil #9: [Request] List available devices

marceloboeira, 2018-05-09:
How about listing the devices within range of your Mac? Its always nice for debugging or even scripting... I thinking about locking my Mac every time my iPhone (over bluetooth) is not in range....

toy, 2018-05-10:
After reading ["Inquiring and Paging"](https://developer.apple.com/library/content/documentation/DeviceDrivers/Conceptual/Bluetooth/BT_Develop_BT_Apps/BT_Develop_BT_Apps.html#//apple_ref/doc/uid/TP30000997-CH216-TPXREF117) in "Bluetooth Device Access Guide" it doesn't look like a good idea to constantly list devices within range. It may be better to connect to a device and track disconnect.

Though device inquiry could still be done for debugging. Also after looking at bluetooth framework docs, much more functionality can be added like listing paired/favourited/recent devices, getting information about device, connecting/disconnecting devices.

twe4ked, 2018-09-05:
I'd love to be able to disconnect certain devices. I need a way to disconnect my bluetooth headphones when my laptop sleeps.

toy, 2018-09-16:
I've added some functionality on [dev branch](https://github.com/toy/blueutil/tree/dev). Not yet sure if it can be considered final.
@marceloboeira Probably calling `blueutil --connect ADDRESS` and doing something on failure exit. Other option is to add functionality for reacting on disconnect notification, but then the connection should be kept open.
@twe4ked `blueutil --disconnect ADDRESS`

twe4ked, 2018-09-16:
Thanks @toy! I'll give it a go.

twe4ked, 2018-09-17:
Works fantastically.

This is how I'm using it:

```
#!/bin/bash -e
#
# https://github.com/toy/blueutil

NAME=QC35
CONNECTION_STRING="$(blueutil --paired | grep $NAME)"
ADDRESS="$(echo "$CONNECTION_STRING" | cut -d' ' -f 2 | sed 's/,//')"

if [[ "$CONNECTION_STRING" = *"not connected"* ]]; then
  blueutil --connect "$ADDRESS"
else
  blueutil --disconnect "$ADDRESS"
fi
```

toy, 2018-09-18:
@twe4ked So you wanted an ability to toggle connection with headphones?

I've already added ability to get information about one device and get connected status and though of ability to find device by name (in the list of recently connected devices)

twe4ked, 2018-09-19:
> So you wanted an ability to toggle connection with headphones?

That was just my test script for now. My plan is to disconnect my headphones when my laptop sleeps and reconnect on wake (this might happen automatically, not sure). Otherwise my headphones stay connected when I'm commuting and have to manually disconnect them to use my iPhone.

commit b9faf731 (2018-10-06): add inquiring devices in range and listing favourite, paired and recent devices, resolves #9
closed 2018-10-06 by toy