toy/blueutil #12: `blueutil -p 0` takes a long time

chdiza, 2018-11-18:
Using `blueutil` to turn the bluetooth power off takes a long time (several seconds).  If memory serves, it used to be basically instantaneous.  Turning bluetooth power on, however, is still basically instantaneous.

This is on High Sierra, with `blueutil` 2.2.0.

toy, 2018-11-18:
Blaming source it doesn't seem anything significant changed in responsible part of code since version 1.1.0. Can you try to compile some earlier version and check if that turns off power faster?

toy, 2018-11-20:
After setting power `blueutil` waits up to 10 seconds for the change to happen with pauses of 0.1s.
Slowness should not be caused by the tool, I'm not sure how to help, maybe simply turning off and on again will help :)

chdiza, 2018-12-06:
I guess there's nothing that can be done.  Turning bluetooth off in System Preferences is instantaneous, but maybe the means to do this instantly are not exposed in their APIs.

toy, 2018-12-06:
You can try watching output of `while true; do blueutil -p; done` while turning off power through System Preferences to see if it is also slow there.

toy, 2018-12-23:
Did you discover if it is `blueutil` responsible for slowness or it is same through System Preferences?

toy, 2019-01-05:
Please reopen if needed

closed 2019-01-05 by toy