toy/blueutil #13: [Request] Pair device

HenriettaSu, 2018-12-19:
Hallo and thanks for that tool. All functionality are pretty!
After reading the code and Apple document, I found maybe a new functionality about pairing device would be added. IOBluetoothDevicePair could be helpful.
My admin locks bluetooth panel, but I need my trackpad....

Thank you.

toy, 2019-01-06:
Have a look at [pair branch](https://github.com/toy/blueutil/compare/pair). I was able to check pairing to devices not requiring authentication, to devices supporting showing pin code, to devices with ability to input pin code and to devices with fixed pin code. I'm still not sure in which case `devicePairingUserPasskeyNotification:passkey:` would be used.

Removing pairing seems to be also possible to implement: https://stackoverflow.com/a/20749542/96823, but it will require either running `blueutil` as super user or granting authorisation through GUI.

HenriettaSu, 2019-01-09:
> Have a look at [pair branch](https://github.com/toy/blueutil/compare/pair). I was able to check pairing to devices not requiring authentication, to devices supporting showing pin code, to devices [...]

Thank you! That's totally awesome!

closed 2019-01-09 by HenriettaSu

toy, 2019-01-10:
@HenriettaSu Did you check if it works for you?