toy/blueutil #14: License

tilmanginzel, 2019-01-19:
Hi @toy,

thanks for continuing the work of Frederik. :) I would like to use the compiled binary in my Alfred 3 [workflow](https://github.com/tilmanginzel/alfred-bluetooth-workflow) and was wondering if you'd have any problems with that.

Do you mind adding a license to this repository so that others and me can be sure that we are allowed to use and distribute the binary with our software (with attribution to this repository)?

Thanks and best regards
Tilman :)

toy, 2019-01-20:
Hi Tilman,

First reaction: Oh no, license question…

Normally I use MIT license for my projects, caring only about attribution and have nothing against integration of compiled binary. This project lacks a license as I'm not sure how to handle license of a public domain fork and you are the first to ask about it ;) Stackexchange [seems to suggest](https://softwareengineering.stackexchange.com/a/307129/8013) to contact Frederik, so I'll try.

toy, 2019-01-21:
@triplef Please have a look at #15 and tell if you want anything to be changed

toy, 2019-01-22:
Any strong opinion on bumping patch, minor or major version?

tilmanginzel, 2019-01-22:
Good question. According to semver it _could_ be seen as a patch, because it's a backwards-compatible change. [Some prefer](https://github.com/semver/semver/issues/322) a major version for license changes though (or even a new project, but this would be a bit excessive...).

IMO you have one of the following options:
* If you think that the license change from none/public domain to MIT _might_ have any negative or breaking impact on current users, be "safe" by using a major version change
* If you think the license change has no negative or breaking impact at all, just use minor or patch

toy, 2019-01-25:
I came to similar conclusion. I feel that change of license doesn't introduce changes requiring major version increase, so I'll go with minor version increase (patch would be not visible enough).

commit ad787d88 (2019-01-22): change license to MIT, resolves #14

with permission from Frederik Seiffert https://github.com/toy/blueutil/issues/14#issuecomment-455985947
closed 2019-01-25 by toy