toy/blueutil #16: MacPorts port / minimum OSX version

DanielO, 2019-02-12:
Hi,
I created a port for this for MacPorts and it works OK, however one issue is they support quite old versions of OSX and this fails to build there.

Do you know the minimum version of OSX that this will build under so I can update the port?

Thanks.

toy, 2019-02-12:
Hi Daniel,
I really appreciate that you've created a port, also as macports user ;)!

As is already written in trac discussion, at least one API is 10.7+.
The xcode project has 10.6 as [deployment target](https://github.com/toy/blueutil/blob/master/blueutil.xcodeproj/project.pbxproj#L216), but that should not be related to building using `make`.

So I've extracted previous SDKs and tried building using xcode and 10.6..10.9 fail with various errors and build succeeds with 10.10 and further ones.
Then by adding `-isysroot path/to/sdk` to `CFLAGS` in `Makefile` I've found that 10.6 still fails by not finding `float.h`, 10.7 and 10.8 build as long as `firstObject` of `NSArray` and `stop` of `IOBluetoothDevicePair` are not used and 10.9 and further build without issues.

So I assume 10.9 should currently be the minimum version.
It should be not hard to bring support at least for 10.7, but I think it is better to do it only if someone will need it.

toy, 2019-02-23:
@DanielO I've made 10.9 minimum version explicitly in e74824825c89dc5a1c665a448ef46d3e2d9d12b5

toy, 2019-03-26:
I assume the issue can be closed
Macports trac ticket: https://trac.macports.org/ticket/58050

closed 2019-03-26 by toy

DanielO, 2019-03-26:
Yes it can be closed, sorry I forgot to tell you!

toy, 2019-04-01:
Thanks again for the port!