toy/blueutil #17: [Request] Configurable (or other than comma) device list field delimiter

ivofrolov, 2019-02-25:
It would be great to have delimiter other than comma in device list for easier parsing with utilities like awk.

For example, comma in second column (`connected (slave, -56 dBm)`) makes it not so easy to split the line with just comma as delimiter.
`address: **-**-**-**-**-**, connected (slave, -56 dBm), not favourite, paired, name: "********", recent access date: 2019-02-25 03:09:43 +0000`

toy, 2019-02-25:
True, this comma makes it not so easy.
I'm wondering what is the common "best" way to solve this, I've seen different options among tools. To try to do it in a good way.

Spaider, 2019-03-26:
I agree, stopped by here just to post the same issue. It violates Unix Way and makes the output not easy to be machine parseable.

toy, 2019-03-26:
@Spaider Suggestions how to do it in a not ugly and decently future proof way while also not jumping to next major version?

Spaider, 2019-03-27:
@toy I don't think it's possible. In any case it will be a breaking change :(
Also, information structure is not consistent. Some of the fields are key-value pairs divided by semicolon and some of them are just flags. I would suggest to create new major version and re-think the way data is present thoroughly.

vitorgalvao, 2019-06-09:
This doesn’t need to be a breaking change. A comma is fine as a separator for human readability but not for machine readability, so we need only introduce a `--machine-readable` or `--alt-separator` flag that outputs information in a different way.

`alt-separator` should be the simpler route. Output the exact same info but use a character that should not appear in the output, such as the pipe character (`|`):

```
address: fd-57-fa-70-b3-35 | connected (master, -70 dBm) | not favourite | paired | name: "My Device" | recent access date: 2019-06-09 18:58:49 +0000
```

One possibility for `machine-readable` is JSON:

```
{
  "fd-57-fa-70-b3-35": {
    "connected": true,
    "connection_info": "master, -70 dBm",
    "favourite": false,
    "paired": true,
    "name": "My Device",
    "recent_access_date": "2019-06-09 18:58:49 +0000"
  }
}
```

toy, 2019-06-10:
I thought that providing output as JSON will require either writing it by hand or adding external library, both of which I would prefer not to do, but just found that there is `NSJSONSerialization` since 10.7, so it is not hard to do. One problem with JSON is that it is easy to parse with a real parser, but not for a simple shell script.
I was thinking of providing an option `--format` with either `json` as value (can also be a separate option `--format-json`), or format string alike git log format string. Then default format will be changed only with the next major release.
There can also be a problem as device name can also contain commas and double quotes.

Spaider, 2019-06-10:
`jq` is used widely to parse JSON files from CLI. So, support JSON output as an option is not a bad idea at all.
@vitorgalvao I think that `connection_info` in your example should be represented as an object because it contains two pieces of data. Something like this (not sure about property names, though ;) :
```
...
  {
    "connection":  {
      "signal": -70,
      "signalUnit": "dBm"
      "mode": "master"
    }
  }
```

toy, 2019-06-11:
@Spaider `jq` is nice, but it will be required to install the tool to parse output. Though JSON output can be a start and another format can come later. Double quotes in name can be a problem only for parsing not JSON output, also if escaped.

toy, 2019-08-05:
@ivofrolov @vitorgalvao @Spaider Any comments on https://github.com/toy/blueutil/compare/formatter?

ivofrolov, 2019-08-07:
Perfect. But I think one option is more than enough. There aren't usually that many device to compact output.

vitorgalvao, 2019-08-07:
> Perfect. But I think one option is more than enough. There aren't usually that many device to compact output.

Agreed. I’d only have added `--json` (or similar) as a binary flag and always output the pretty print version.

Still, I don’t think it’s a big deal. Eager for the new version release so I can use it.

Thank you.

toy, 2019-08-10:
@ivofrolov I agree that it is hard to get long list of devices and comparing size of output the pretty version is only 1/3 bigger than compact version. I just prefer the default json output to not be wasteful, as it will anyway be consumed by some tool, though give the convenience of easily readable output without any additional tools for humans, especially as amount of code for allowing both is very small.
@vitorgalvao I was thinking of `--json`, but decided to stick with `--format=*` to make it easy to extend with whatever other formatter (named like `csv`, `tsv` or just `%a,%b,%c`).
I'm now thinking that it may be good to introduce some new default output right away.
Does any one of you have any comment on json schema, key names, …?
@ivofrolov Can it be considered that the changes in the branch resolve the issue?

ivofrolov, 2019-08-11:
> Can it be considered that the changes in the branch resolve the issue?

Yes!

toy, 2019-08-16:
I've also added `new-default` format to replace default with next major version change. Any comments about it?

Listing multiple devices:
```
address: aa-bb-cc-dd-ee-ff, recent access: 2019-08-16 00:00:00 +0000, favourite: no, paired: yes, connected: master, rssi: 0, raw rssi: -45, name: Some foo
address: aa-bb-cc-dd-ee-ff, recent access: 2019-08-15 00:00:00 +0000, favourite: no, paired: yes, connected: slave, rssi: -7, raw rssi: -98, name: Some bar
address: aa-bb-cc-dd-ee-ff, recent access: 2019-08-14 00:00:00 +0000, favourite: no, paired: no, connected: no, rssi: -, raw rssi: -, name: Some baz
```

Info about one device:
```
address: aa-bb-cc-ss-ee-ff
recent access: 2019-08-14 00:00:00 +0000
favourite: no
paired: no
connected: no
rssi: -
raw rssi: -
name: Some name
```

vitorgalvao, 2019-08-20:
@toy I’ll only ever use `--format json` once that is released, but `new-default` looks fine to me.

toy, 2019-08-21:
@vitorgalvao Thanks for feedback, I'll release new version then.

toy, 2019-08-21:
v2.5.0 is out

commit 32d7b079 (2019-08-16): allow setting formatter to json/json-pretty, resolves #17
commit d0fc9447 (2019-08-17): allow setting formatter to json/json-pretty, resolves #17
commit 3b002e33 (2019-08-21): allow setting formatter to json/json-pretty, resolves #17
closed 2019-08-21 by toy