toy/blueutil #18: blueutil command doesn't work at all in a "do shell script"

jb365, 2019-03-04:
Hello, I'm having trouble using blueutil in a "do shell script" in AppleScript, or any use of a shell script in Automator actually. I got the binary from Homebrew if that helps.
Hope you have some advice!

toy, 2019-03-04:
You didn't describe the trouble you are having.

I checked the basics and didn't find any problems, except maybe common problem with environment been different from what you'd get in terminal.
If `blueutil` is in your `PATH` (`do shell script "env | grep PATH"`), then you should be able to call it as is like `do shell script "blueutil -p -d"`.
Otherwise you will need to specify full path `do shell script "/full/path/to/blueutil -p -d"`.

jb365, 2019-03-04:
I ran "blueutil -p" in terminal, outputs 0 like normally. But after running do shell script "blueutil -p" in AppleScript it outputs nothing at all, no errors either. I have tried using the full path as well.

toy, 2019-03-05:
I assume you are using the "Run AppleScript" action, did you check the "Results" of the action?
I see `0` or `1` there for `do shell script "blueutil -p"`

jb365, 2019-03-15:
Sorry I forgot to reply, but I never got a 0 or 1, it completed the script successfully but doesn't output anything.

toy, 2019-03-15:
You need to provide more information on what exactly do you do, to get more suggestions.

toy, 2019-03-26:
@jb365 Do you still want to continue with this issue?

jb365, 2019-03-26:
I would like to, but I don’t know how else to describe it. I’m literally in Automator running `do shell script “blueutil -p”`, it compiles and runs fine, but nothing outputs at all. Running the command in terminal works fine however.

toy, 2019-03-26:
I've got a hunch after once more retrying to create the workflow: do you run the complete workflow or just press play button on "Run AppleScript" action? Running the complete workflow produces result of the action and whole workflow, or fails with an error (also as a modal dialog), but pressing play seems to run the script without capturing result (though it will show if there is an error).

jb365, 2019-03-26:
I’ve been just clicking the play button, I’ll try completing it instead. My main problem was with BetterTouchTool however, as running the AppleScript option wouldn’t return anything. Unless there is a way to run a a workflow from AppleScript itself?

toy, 2019-03-26:
You mean there is a possibility to run AppleScript from BetterTouchTool? Then you can run AppleScript directly, no need to wrap it in a workflow.

jb365, 2019-03-26:
That’s the problem, the run AppleScript in BetterTouchTool returns nothing (just like the play button in Automator). So maybe it would work if it ran a workflow instead?

toy, 2019-03-26:
Are you sure BetterTouchTool is supposed to do something with the result of last command in the script? Did you try running some other tool like `do shell script "env"`, do you see output anywhere?
If you just want to debug the result of blueutil when output is not show, you can use `display dialog` like `display dialog (do shell script "blueutil -p")` or wrap in `try … on error …` to check for errors.

jb365, 2019-03-26:
In BetterTouchTool I get `sh: blueutil: command not found`. It gets found fine in terminal though.

toy, 2019-03-31:
You need to specify full path to blueutil or change environment, environment for gui applications is changed differently from how it is done for ones shell, have a look at  https://stackoverflow.com/q/135688/96823

jb365, 2019-03-31:
Ok, what environment should I be in then? I googled around a little but couldn't find anything helpful.

toy, 2019-04-01:
I was not clear enough in my answer, by environment I meant environment variables, specifically `PATH`, have a look at the answer on stackoverflow at the link I gave. Changing environment is only needed if specifying full path to `blueutil` is not an option.

vitorgalvao, 2019-06-09:
@jb365 If you’re still having problems, and since you installed with Homebrew, the solution is as simple as using `/usr/local/bin/blueutil` instead of `blueutil`, in your script.

toy, 2019-08-10:
@jb365 Would you like to continue discussing your issue?

jb365, 2019-08-10:
@toy I kinda just gave up, nothing I did would fix it. The problem probably came from using a something other than terminal I guess.

toy, 2019-08-10:
Did you try suggestion from @vitorgalvao?

jb365, 2019-08-10:
Yes I tried that a while back.

toy, 2019-08-10:
I can't imagine a good reason for a tool working from Terminal to not work from Automator, so if you would like to try to make it work, I will try to help.

jb365, 2019-08-10:
Ok that’d be great, are you available tomorrow? Also do you have someplace easier to communicate, like discord, slack, or twitter?

toy, 2019-08-10:
I'd stick for now to asynchronous communication through this issue, also checking this issue can help someone else having similar problem.
To start you need to find out the location of `blueutil` binary. Try `which blueutil` in your terminal and please post the result. If you installed through homebrew and didn't change install location, it should be `/usr/local/bin/blueutil`.

toy, 2019-08-21:
Closing due to absence of reply. Please reopen or open a new issue if you would like to discuss the issue.

closed 2019-08-21 by toy

AntoLepejian, 2024-01-30:
I just ran into this exact same issue, the problem was that in automaton, your `.zshrc` or any other bash profile isn't loaded, and as such your `$PATH` is missing a lot of executables.

To remedy it, I just added this one line before my script

```bash
# this line sets up my environment, without it $PATH isn't set, and as such most executables won't work in automator
source ~/.zshrc

blueutil -p toggle
```