toy/blueutil #20: not working with wiimote

SevenChords, 2019-04-14:
I try to connect my wiimote via bluetooth, which worked fine before macos forced a bluetooth pin to connect, but the wiimotes don't have a bluetooth pin, and since dolphin could connect it (wasn't usable outside of dolphin), i imagined this could too, but appearently this can't connect to devices that simply don't have a bluetooth pin either, was wondering if this is a bug with the current version of this tool or if it has become literally impossible to connect bluetooth devices that don't have a bluetooth pin.

toy, 2019-04-19:
By dolphin you mean https://dolphin-emu.org/?
Do you get output from the tool and can you share it?
I don't have a wiimote, but when developing the pairing functionality I was testing it on different devices including one without a pin and one with a default pin.

toy, 2019-08-10:
@SevenChords Would you like to continue discussing your issue?

toy, 2019-08-21:
Closing due to absence of reply. Please reopen or open a new issue if you would like to discuss the issue.

closed 2019-08-21 by toy

gingerbeardman, 2023-03-20:
This is still not working, I'm on Monterey and it seems according to other issues it is an unresolved macOS bug.

Curious that [Wii4SESlogger](https://stemed.online/wiimote/) which uses `blueutil` behind the scenes to pair and connect shows a screenshot that implies it works on macOS Monterey.

[image]