toy/blueutil #21: 'Failed to connect' error on recognized Apple Magic Trackpad 2

dhowe, 2019-06-06:
```
$ blueutil --inquiry
address: ec-2c-e2-ef-44-0f, not connected, not favourite, not paired, name: "Magic Trackpad 2", recent access date: (null)

$ blueutil --connect ec-2c-e2-ef-44-0f
Failed to connect "ec-2c-e2-ef-44-0f"

```

toy, 2019-06-10:
From the output of inquiry, it doesn't seem to be recognised. Did you try connecting through UI?

toy, 2019-08-10:
@dhowe Would you like to continue discussing your issue?

dhowe, 2019-08-13:
Sorry, don't have access to a trackpad anymore, but yes, I could connect consistently through the UI, but not from the command line

toy, 2019-08-15:
Unfortunately I don't see a way to debug this issue unless you get access back or someone else comes with the same problem. I've tried unpairing and initiating pairing, and multiple rounds of connecting/disconnecting on previous version of trackpad, but I don't have access to the new one.

closed 2019-08-15 by toy