toy/blueutil #22: Explain keyboard shortcut creation

MrMinimal64, 2019-07-13:
I guess many users want to use this package to create a keyboard shortcut for automatically switching bluetooth on and off.
Why not include the instructions for doing so in the readme (or somewhere else)?
Something like:

> To create a keyboard shortcut that toggles bluetooth:
> Install blueutil
> Start Automator, and create a new Quick Action (Service).
> Set "Service receives selected: to "no input" in "any application".
> Add an action named "Run Shell Script". It's in the Utilities section of the Actions Library.
> Insert this bash command into the text box:
> 
> ` 	/usr/local/bin/blueutil -p toggle`
> 
> Test run it using the Run button (top right).
> Save it, giving it a service name you can remember.
> Go to System Preferences -> Keyboard, and go to the Shortcuts tab
> Go to the Services section, and scroll down to General - you should find your service there. If you select the line, you can click "add shortcut" and give it a keyboard shortcut.

toy, 2019-07-14:
Interesting idea. I'm a bit hesitant about readme file itself (it is a CLI tool), but there is wiki on github or an FAQ or "usage ideas" file can be added and linked from readme. What do you think?

MrMinimal64, 2019-07-15:
all sound fine. perhaps one could reference the "usage ideas" file in the wiki and FAQ (use all 3 options). My solution above only includes MacOS by the way. Not sure how to achieve this with other OS.

toy, 2019-08-10:
Maybe the simplest way is to start with a wiki page, would you like to create it?

MrMinimal64, 2019-08-20:
I just did: https://github.com/toy/blueutil/wiki
Unfortunately I am not able to put more work into in right now.

closed 2019-08-20 by MrMinimal64

toy, 2019-08-21:
Thanks!
Please tell why did you add Linux/Windows sections? `blueutil` is not cross-platform and there is no plan to make it cross-platform.

MrMinimal64, 2019-08-21:
fixed it. of course that did not make sense in the wiki of this package!