toy/blueutil #23: MacOS Notarization

tilmanginzel, 2019-08-03:
Hey @toy,

since MacOS Catalina (currently in beta), all applications have to be notarized by default. Otherwise, Gatekeeper won't allow to open the app. See:

- https://developer.apple.com/news/?id=06032019i
- https://developer.apple.com/documentation/security/notarizing_your_app_before_distribution
- https://eclecticlight.co/2019/06/07/notarization-in-mojave-and-catalina/

Do you have a developer ID certificate? If yes, would it be possible for you to notarize the app, so it will run without any issues in future MacOS versions?

Thanks in advance!

Cheers
Tilman

Reference: https://github.com/tilmanginzel/alfred-bluetooth-workflow/issues/9#issuecomment-517847262

toy, 2019-08-03:
Hi Tilman. If I understand correctly notarisation is done for binaries, but the project itself is currently distributed only as source files. From the articles it seems that it should be possible to use a binary compiled on local system. And code-signing/notarisation of homebrew binaries seems to be on the roadmap - https://discourse.brew.sh/t/code-signing-installed-executables/2131/9.

tilmanginzel, 2019-08-04:
Hi,

> but the project itself is currently distributed only as source files

Ah good point. I forgot that I have built blueutil via `make` and distributed that binary with my workflow. This makes the problem more complicated for me. Notarization is completely knew to me so I am still not entirely sure how I should proceed with this:

* Notarize the blueutil binary I have built myself? It would feel weird to codesign and notarize someone elses applications, Apple staff also [discourages this](https://forums.developer.apple.com/thread/120223)
* Notarize only my workflow bundle, which contains the blueutil binary? Would that be enough?
* Force users to install all dependent binaries themselves? I do not think that this is the way it's intended by Apple

In the end users should always be able to bypass Gatekeeper for certain applications, so this will likely work as fallback.

toy, 2019-08-10:
The suggestion at the link is for the binary to be notarise by the same person that code-signs it, so no problem doing both by the same person. Though I understand why it can feel strange :)
Also check if notarising the workflow is enough, maybe it is the solution.

For me distributing only source is much easier, it should also be easy to build or install it using homebrew, and I assume that tool is directly used mostly by people for whom terminal is not an unknown tool. Though I understand that Alfred users can simply not know terminal or how to compile code, so if first two options that you proposed don't work, please reopen this issue or open a new one.

closed 2019-08-10 by toy

tilmanginzel, 2019-08-11:
Yes that sounds reasonable. I have put this issue on halt for me too until MacOS Catalina is released. I will keep you updated if I find a solution. :)