toy/blueutil #24: `(null)` values crash blueutil when asking for JSON format

vitorgalvao, 2019-08-24:
When running `blueutil --paired`, one of my devices (a keyboard I only use on occasion) shows itself as:

```console
address: 00-1f-20-4d-4f-a9, not connected, not favourite, paired, name: "(null)", recent access date: (null)
```

That causes problems with the new JSON format:

```console
$ blueutil --format json --paired

2019-08-24 23:21:57.329 blueutil[24636:830763] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '*** -[__NSPlaceholderDictionary initWithObjects:forKeys:count:]: attempt to insert nil object from objects[1]'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fff4cd5e2fd __exceptionPreprocess + 256
	1   libobjc.A.dylib                     0x00007fff7742fa17 objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff4cd9d2f4 -[CFPrefsConfigurationFileSource initWithConfigurationPropertyList:containingPreferences:] + 0
	3   CoreFoundation                      0x00007fff4cc763eb -[__NSPlaceholderDictionary initWithObjects:forKeys:count:] + 311
	4   CoreFoundation                      0x00007fff4ccb17ee +[NSDictionary dictionaryWithObjects:forKeys:count:] + 59
	5   blueutil                            0x00000001094dd64a list_devices_json + 967
	6   blueutil                            0x00000001094de4a6 main + 1904
	7   libdyld.dylib                       0x00007fff78bfd3d5 start + 1
	8   ???                                 0x0000000000000004 0x0 + 4
)
libc++abi.dylib: terminating with uncaught exception of type NSException
zsh: abort      blueutil --format json-pretty --paired
```

Unpairing the device removes the error.

toy, 2019-08-25:
Thanks for reporting, it seems it should not be hard to fix.

toy, 2019-08-25:
https://github.com/toy/blueutil/compare/fix-for-null-name-or-access-date should fix the problem, please tell my if you think that - for null name and recent access date in default and new default formats looks wrong.

vitorgalvao, 2019-08-26:
Because I’ve unpaired the problematic device (to test if that was indeed the cause), I can no longer test (readding the keyboard now is not giving any `(null)` values.

closed 2019-08-27 by toy

vitorgalvao, 2019-08-27:
Thank you for the fast fixes!

toy, 2019-08-27:
Unfortunate that you can't test it, I was partially able to test it by inquiring devices and got one which was never accessed.
Release this and #25 in [v2.5.1](https://github.com/toy/blueutil/releases/tag/v2.5.1)

commit a06b3163 (2019-08-27): handle null for name and recent access date for all formats, resolves #24