toy/blueutil #25: The argument position of `--format` matters

vitorgalvao, 2019-08-24:
Thank you again for the new format options. One thing that tripped me up is that `--format` needs to come before the other options. `blueutil --format json --paired` works but `blueutil --paired --format json` does not (it shows the `default` format).

Ideally, `--format` could be invoked in any position, but if that’s not possible `help` should mention it needs to come first.

toy, 2019-08-25:
This is caused by how arguments are handled starting with version 2.0.0 – it is possible to run multiple commands which makes most sense for something like `blueutil --power 1 --discoverable 1` (though can also be done with two successive commands). I'm not really sure if it is useful, it is probably there mostly because nothing was preventing from doing it that way.

`--format` is the first switch that affects global state and it is clear that it may be confusing that it affects only commands following after it. I doubt that it makes a lot of sense to allow changing output format between commands, so probably it is better to consider it a bug and use last specified format for all commands.

toy, 2019-08-25:
https://github.com/toy/blueutil/compare/set-format-only-once should remove the confusing behaviour

commit c28860f8 (2019-08-27): use last specified format for all output, resolves #25
closed 2019-08-27 by toy