toy/blueutil #26: Compile to 64bit?

Mortennn, 2019-09-21:
I noticed Blueutil is compiled in 32bit which means it won't work on macOS Catalina. Would it be possible to release a 64bit build?

toy, 2019-09-21:
Hi Morten.
The project itself is currently distributed only as source files, this was touched a bit more in #23.
I've checked what should one get from macports and brew, and both should give you 64 bit binary (at least for 10.14).
I've also checked the project configuration file and there is nothing about architectures, it was cleaned up as one of first commits c0fc5e480370a5b09a0850a66b400226c80c70ba, so unless one explicitly compiles for i386, he should be getting a 64 bit executable.
Do you compile yourself or do you get the binary from somewhere?

Mortennn, 2019-09-21:
[Go64](https://www.stclairsoft.com/Go64/index.html) detected Blueutil as a 32 bit binary, but it must be a false positive then.

Thank you for your answer.

closed 2019-09-21 by Mortennn

toy, 2019-09-21:
Could you please check the binary you have using `file`, so running `file /path/to/blueutil` (`file $(which blueutil)`)?
Weird how could they detect it wrong, it is not even an app bundle with possibility of multiple binaries inside.

Mortennn, 2019-09-21:
I was actually using a 32 bit binary.
```
/usr/local/bin/blueutil: Mach-O universal binary with 2 architectures: [ppc_7400:Mach-O executable ppc_7400] [i386:Mach-O executable i386]
/usr/local/bin/blueutil (for architecture ppc7400):	Mach-O executable ppc_7400
/usr/local/bin/blueutil (for architecture i386):	Mach-O executable i386
```

But after overwriting a symlink I got the 64 bit binary.
```
/usr/local/bin/blueutil: Mach-O 64-bit executable x86_64
```

toy, 2019-09-22:
Yes, looks like you had some old binary installed on your system