toy/blueutil #27: one important option is missing - Unpair Paired Bluetooth Device

deliad2, 2019-09-28:
because I'm using Bluetooth wireless keyboard with 2 computers every time I want to connect to the other one, I have to repair the keyboard with the computer Bluetooth

so I have to unpair the device from OSX first and pair it again in order to reconnect it to OSX after using the keyboard in another computer

toy, 2019-09-29:
Unfortunately there is not public interface in osx bluetooth framework for removing pairing. I remember reading somewhere that inability to remove pairing is a security feature. I tried few times to find how it is done by the bluetooth menu item or bluetooth preference pane, but did not succeed and help is very welcome.

Though I'm wondering why do you need to unpair device to connect it to other computer, shouldn't it be enough to disconnect it from first and connect to the second one?

toy, 2020-02-16:
@deliad2 Please reopen if needed

closed 2020-02-16 by toy

rikukissa, 2021-03-19:
That's unfortunate. I'm also trying to create some automation for switching my magic mouse and keyboard between 2 macs. 

>Though I'm wondering why do you need to unpair device to connect it to other computer, shouldn't it be enough to disconnect it from first and connect to the second one?

"The trouble is with the devices, the Apple mouse and the keyboard in your case. They don't pair with more than one BlueTooth host at a time. You need a BlueTooth host that's "central" to both Macs if you want to use them."
https://apple.stackexchange.com/a/23154

toy, 2021-03-19:
@rikukissa Please check https://github.com/toy/blueutil/issues/46#issuecomment-778689350 if you goal is switching mouse/keyboard between 2 macs. But I'll anyway still try to add unpairing when time allows.

akashagarwal7, 2021-04-05:
+1, same troubles here!

timberhill, 2021-04-15:
I have a slightly different problem that would be solved by unpairing a bluetooth device. I use a Logitech G603 and there is a button to switch between bluetooth connection (that I use on Mac) and a dongle (that I use on a PC). The switch worked very well the first time, but after a restart, whenever I power up the Mac or switch to the dongle and back the bluetooth mouse connects, but doesn't respond until unpaired and paired again.

Just wondering if anyone has any ideas about this, I realise that unpairing might not be possible via a terminal.

toy, 2021-05-23:
@deliad2 @rikukissa @akashagarwal7 @timberhill @falconws @episourcedavid I've stumbled upon an undeclared method allowing unpairing functionality, it is available with just released v2.9.0, details in [comment for #46](https://github.com/toy/blueutil/issues/46#issuecomment-832326520).

commit 889250eb (2021-05-05): add unpairing functionality, resolves #27, resolves #31, resolves #32, resolves #46, resolves #53
commit 1a5a5c61 (2021-05-05): add unpairing functionality, resolves #27, resolves #31, resolves #32, resolves #46, resolves #53

Kudos to @JanC for suggesting how to avoid crash if the undeclared
method that is used is not available.
commit 2921fa98 (2021-05-05): add unpairing functionality, resolves #27, resolves #31, resolves #32, resolves #46, resolves #53

Kudos to @JanC for suggesting how to avoid crash if the undeclared
method that is used is not available.
commit 05ce3862 (2021-05-23): add unpairing functionality, resolves #27, resolves #31, resolves #32, resolves #46

Kudos to @JanC for suggesting how to avoid crash if the undeclared
method that is used is not available.