toy/blueutil #30: It is not working on MAC OS Catalina

swordfishub, 2020-04-02:
When I run blueutil on MAC OS Catalina is throwing the following error:
zsh: bad CPU type in executable: blueutil
It seems that Catalina doesn't support 32 bits applications:
https://github.com/electron/node-rcedit/issues/51
Could you update the application to 64 bits?

toy, 2020-04-02:
This repository contains only source code of the tool, binaries come from [homebrew](https://formulae.brew.sh/formula/blueutil), [macports](https://github.com/macports/macports-ports/blob/master/sysutils/blueutil/Portfile) or some other place, please have a look at #26 and #23, or instructions on how to [build from source](https://github.com/toy/blueutil#from-source)

arvenil, 2020-04-03:
Just installed it on osx from homebrew and it works.

lkalneus, 2020-04-14:
disconnect works fine for me, but I can't connect to the device back :(

toy, 2020-04-26:
@lkalneus I assume it is a separate issue, so would be better to open a new one. Is the device you are trying to reconnect to is still turned on?

toy, 2020-05-09:
@swordfishub Please reopen if needed
@lkalneus Please open a separate issue

closed 2020-05-09 by toy