toy/blueutil #31: Can I unpair the device with this tool?

arvenil, 2020-04-03:
I can `--pair` the device, but I don't see how can I unpair it?

closed 2020-04-03 by arvenil

toy, 2020-04-04:
Unfortunately there is no interface and it seems to be possible only as superuser, have a look at #27 and https://stackoverflow.com/a/20749542/96823

arvenil, 2020-04-04:
Hi, thanks for the reply. Yes, I closed this issue because I saw #27.
Actually https://stackoverflow.com/a/20749542/96823 gives a hope. I
wouldn't mind if this program would require sudo for removing device.
However I have a tiny issue so I'm not sure if this is worth the effort. So
for now I'm stuck to old method of going to menu, removing the device and
then now I re add it from cmd line which is quite helpful. The issue I have
is with headphones and sharing osx with win10 (bootcamp). If I pair the
device in win10, they won't work in osx, If I pair in osx, they won't work
in win10. Known bug, probably related with the fact that the id of
bluetooth is the same in both OS, but database of paired devices is not
shared.
https://github.com/Soorma07/OS-X-Bluetooth-Pairing-Value-To-Windows-Value
(but this is not working anymore).

toy, 2020-04-04:
The issue you have is a peculiar one :) The code on stackoverflow gives hope indeed, but as it is only changing the plist, I assume that the bluetooth daemon should be informed to reload (and I doubt that it is possible) or restarted. As I already commented in #27 I wasn't yet successful in digging into bluetooth menu item and preference panes.

toy, 2021-05-23:
@arvenil I've stumbled upon an undeclared method allowing unpairing functionality, it is available with just released v2.9.0, details in [comment for #46](https://github.com/toy/blueutil/issues/46#issuecomment-832326520).

commit 889250eb (2021-05-05): add unpairing functionality, resolves #27, resolves #31, resolves #32, resolves #46, resolves #53
commit 1a5a5c61 (2021-05-05): add unpairing functionality, resolves #27, resolves #31, resolves #32, resolves #46, resolves #53

Kudos to @JanC for suggesting how to avoid crash if the undeclared
method that is used is not available.
commit 2921fa98 (2021-05-05): add unpairing functionality, resolves #27, resolves #31, resolves #32, resolves #46, resolves #53

Kudos to @JanC for suggesting how to avoid crash if the undeclared
method that is used is not available.
commit 05ce3862 (2021-05-23): add unpairing functionality, resolves #27, resolves #31, resolves #32, resolves #46

Kudos to @JanC for suggesting how to avoid crash if the undeclared
method that is used is not available.