toy/blueutil #33: How to change the disconnect timeout of bluetooth keyboard

Tasan007, 2020-05-22:
My bluetooth keyboard is disconnecting automatically after 5 minutes of inactivity on my MacBook air 2019. How can I extend the timer of 5 minutes for this device with blueutil. 

I have already tried the following, but it doesn't work for me.

blueutil --wait-disconnect [my BT ID] 999999999999999

toy, 2020-05-23:
My guess is that the disconnection is caused by the keyboard and not by the laptop, so I don't think that you can configure it on laptop side. Could you share if it is an Apple keyboard?

To try to investigate why disconnect wait is not triggered, could you post here output of `blueutil --info [your BT ID] --format json` while connected and after disconnect (I don't think that address and name are valuable to debug this, so no need to post them if you don't want to). You can run a loop like `for n in {1..60}; do sleep 10; blueutil --info [your BT ID] --format json; done` to show output for 10 minutes every 10 seconds.

Tasan007, 2020-05-23:
Disconnection is not caused by the keyboard, because it's not losing connecting when paired with my phone or with my windows laptop. I have the same issue with my mouse which is also automatically disconnecting on my mac after 5 min. inactivity, but it reconnects automatically when moving the mouse. On my keyboard I have to push the reconnect button every time. Hitting any button on the keyboard after disconnection doesn't lead to auto reconnect.

This is the output of blueutil --info [your BT ID] --format json.

While connected:

{"address":"54-46-6b-f1-12-69","recentAccessDate":"2020-05-23T18:10:55+02:00","paired":true,"RSSI":0,"rawRSSI":-55,"favourite":false,"connected":true,"name":"Bluetooth 3.0 keyboard ","slave":false}%  

After disconnection:

{"address":"54-46-6b-f1-12-69","recentAccessDate":"2020-05-23T18:10:55+02:00","favourite":false,"name":"Bluetooth 3.0 keyboard ","connected":false,"paired":true}%

toy, 2020-05-23:
So `blueutil --wait-disconnect [your BT ID]` doesn't return when disconnection happens? And if it does, does `blueutil --wait-disconnect [your BT ID] --connect [your BT ID]` reconnect the keyboard?

Tasan007, 2020-05-23:
blueutil --wait-disconnect [your BT ID] just returns when disconnected I think. So it looks like you can execute this command. What do you normally needs to see when hitting this command in terminal? I don't get any error message or other message in terminal.

--connect [your BT ID] gives the following error message:
Failed to connect "54-46-6b-f1-12-69"

toy, 2020-05-25:
The `--wait-disconnect` is just waiting for disconnection, output will be only on error.
The reconnection can't be initiated from the laptop side, but I suggest for certainty to try to reconnect the keyboard from system preference pane or menu bar item, though they are probably using the same API.

Sadly I don't know of any setting in macOS bluetooth related to timeout, but the keyboard is working without problems with other devices, so I highly suggest trying something for [similar problems](https://apple.stackexchange.com/search?q=macos+bluetooth+disconnect).

toy, 2020-08-16:
@Tasan007 Did you have any luck with timeout problem?

toy, 2021-03-21:
@Tasan007 Please reopen if needed

closed 2021-03-21 by toy