toy/blueutil #34: battery display

alanhg, 2020-06-01:
I've noticed that if it's an apple device, you can see the power in a Bluetooth-managed location.

Can I get power information for these devices?

toy, 2020-06-01:
There is no such functionality in `blueutil`. There is `ioreg` which is used at the link you gave and its functionality can probably be reproduced using IOKit, but not sure if non apple devices are giving information. There is also [Akku app](https://github.com/jariz/Akku) which I assume is more universal, but looking at ammount of code it is apparently is pretty complex.

alanhg, 2020-06-01:
@toy 

Thank you, I'll try ioreg

alanhg, 2020-06-01:
I found solution 

 https://apple.stackexchange.com/questions/328091/airpods-battery-level-with-ioreg

closed 2020-06-01 by alanhg