toy/blueutil #36: Consider CLI ability to perform HCI Controller Selection

artcashin, 2020-06-12:
For those of us with older hardware, it would be great to have the ability to turn off the internal Bluetooth and turn on an USB dongle that supports 4.0 or 5.0. Similar to the HCI Controller Selector in Bluetooth Explorer.

We can modify the startup to inventory the connected devices, stop the built-in controller, recognize the dongle and reconnect the devices.

toy, 2020-06-13:
First thank you for mentioning Bluetooth Explorer, didn't know about it :)
Besides GUI it includes `BluetoothHardwareSwitchTool` which apparently should switch controllers, but I can't check functionality as I don't have a USB dongle.

Overall I assume the interface to be somewhere in IOKit, but I didn't find it after searching for some time.

antonyupward, 2020-07-21:
Thanks for this comment on BluetoothHardwareSwitchTool.  Very useful... been looking for this command line functionality for ages.

Probably a noob piece of additional info obvious to most... this tool is not in
`/Library/Developer/CommandLineTools/usr/bin` 
along with all the other XCode tools...

It is stored *inside* the Bluetooth Explorer app package... 
`/Applications/Utilities/Bluetooth Explorer.app/Contents/Resources/BluetoothHardwareSwitchTool
`
There are a couple of other command line tools there too... but not had time to investigate.

Possible Humorous Aisde: I'm amazed / not surprised that the BluetoothHarwareSwitchTool is not documented / mentioned anywhere else on the web as per Google.  Indeed the only result to search for BluetoothHardwareSwitchTool is @toy 's post above!  Its almost a [googlewhack](http://www.otherbb.com/2015/09/google-searches-that-give-a-single-result.html) :-)

toy, 2020-08-27:
@artcashin Finally got my hands on a USB dongle and was able to check functionality.
Was able to successfully switch controller through "Bluetooth Explorer", not good that it requires elevated privileges as it means that it would be required for the tool too. Also accidentally caused a system crash! when tried to switch controller after removing the dongle.
Unfortunately was not able to do anything using `BluetoothHardwareSwitchTool` it even didn't list USB dongle using `-p` switch.

Looking at answers for https://apple.stackexchange.com/q/94402/2084 I was able to switch to the dongle by connecting it when bluetooth was off. And it reconnected to the dongle after restart. Also see https://developer.apple.com/library/archive/technotes/tn2295/_index.html#//apple_ref/doc/uid/DTS40012211-CH1-UNDERSTAND_SWITCH__BEHAVIOR.
So it looks like if USB dongle should be always used instead of the build-in controller, then functionality is already there in the OS.

toy, 2021-03-21:
@artcashin Please reopen if needed

closed 2021-03-21 by toy