toy/blueutil #37: Release Tarballs

beren12, 2020-06-26:
Could you please issue release tarballs so it's easier for people to download a specific version of the source? It can also be checksummed/signed so there is no doubt it hasn't been modified

toy, 2020-06-29:
Tarballs for releases are already there https://github.com/toy/blueutil/releases and after some searching it seems that unlike archives linked at https://github.com/toy/blueutil/tags they are stable.
But checksums should be easy to add to releases, not sure whether putting them in description or as files like `blueutil-2.6.0.tar.gz.sha256` is better. Signatures on the other hand are more involved.

beren12, 2020-06-30:
Sorry, looks like github's new layout got me. So much for an improvement… I was looking for them at the top not hidden in the sidebar. Checksums are likely good enough in the description if it's not too much work.

toy, 2020-06-30:
You are not alone, it took me some time to find the link 😄 
Created a release out of latest tag: https://github.com/toy/blueutil/releases/tag/v2.6.0, I'll see how to better organise checking and checksumming archives to add description in future releases

beren12, 2020-06-30:
Oh. I just saw firefox had this repo open... there was *no* releases link anywhere when I opened the bug, not until I refreshed just now and it appeared. so I *wasn't* crazy/blind.

toy, 2020-08-16:
Created [releases for each tag with checksums](https://github.com/toy/blueutil/releases), so I assume this can be closed :)

commit 9fdbf22a (2021-03-22): script to verify release and output checksums, resolves #37
closed 2020-08-16 by toy