toy/blueutil #38: Unable to connect after using --disconnect

JJGO, 2020-06-30:
First, I'm on macOS Catalina 10.15.4

I issued a --disconnect command that successfully disconnected my Bluetooth headphones. I'm now unable to connect them back.

I have tried `blueutil connect`, unpairing and repairing the headphones, resetting the Bluetooth module and every time I try to connect the headphones they show as connected for a couple of seconds and then quickly disconnects.

How can I undo whatever `blueutil --disconnect` did? I can't use my headphones anymore.

toy, 2020-06-30:
Disconnect command just calls the method to close the connection on selected device, so inability to reconnect is odd.
Could you try once more disconnecting and reconnecting using `blueutil`, and the same using bluetooth menu item or through preference pane? Also if possible with some other mac?

toy, 2020-08-16:
@JJGO Your input is needed

JJGO, 2020-08-16:
I'm afraid that after all the pain it was to reconnect the pair of headphones after the disconnect, I gave up on using the software. I do not have other mac to test this on unfortunately. For reference, the headphones were Sony WH-1000X-M3. Feel free to close the issue.

toy, 2020-08-16:
Sad to hear that. Would it still be possible for you to try if disconnecting using bluetooth menu item or through preference pane doesn't cause same issue?

JJGO, 2020-08-16:
Yes, I connect and disconnect the headphones a couple of times a day with the System Bluetooth prefpane or menu item without any issue.

toy, 2020-08-16:
Were you not able to connect headphones back the first time you disconnected them using `blueutil`? Also did you try it again or abandoned the tool after restarting?

mpkasp, 2020-12-02:
Hi @toy I'm having the same issue as @JJGO . If I call disconnect and then connect immediately, everything works great. I see more issues when going between my two laptops using my Apple Magic Keyboard. It eventually disappears from my bluetooth list and I need to switch it off and on for it to be discovered by the OS again.

I'll see if I can get some reproducible steps for you if that will help.

mpkasp, 2020-12-04:
Ok while doing additional testing I ended up taking a few steps back until finally just manually connecting and disconnecting from the Mac OS bluetooth menu. The issue seems to be reproducible using the menu so it appears that at least my use case may not be a blueutil issue.

Leaving another observation in case someone else lands here - 
1) Macbook #1: I use the menu to disconnect my Magic Keyboard, then close Macbook #1 to put it to sleep.
2) Macbook #2: Now I try to connect on a Macbook #2, and the connection will fail (with no error). The bluetooth device then disappears from the connection list in the menu.
3) Back to Macbook #1: I then go back and open Macbook #1, and the keyboard is re-connected.

I even tested disabling "Allow Bluetooth devices to wake this computer" on both laptops to no avail. 

Thanks for putting together a cool library, @toy !

toy, 2020-12-13:
@mpkasp Thanks for reproducing it from bluetooth menu. I will also try checking this behaviour, but your experiment puts it is on framework/kext/os level.

closed 2021-03-21 by toy