toy/blueutil #41: Setting Discoverable as Root Disables Discoverability for the System

Brikwerk, 2020-10-26:
## The Problem

Using blueutil as a root user and setting discoverability causes the system to lose the ability to be discovered over Bluetooth.

## Replicating the Issue

To replicate this bug (**which I advise against replicating since I haven't been able to reverse it**), run the following:

```bash
sudo blueutil -d 1
```

## Attempted Debugging

I've tested a couple different configurations to double check that one device wasn't simply refusing to acknowledge my Mac. I'll list out a few things I tried:
1. Opening the Bluetooth Preferences menu and checking if the Mac was discoverable on an iPhone, iPad, Linux laptop, and a Raspberry Pi.
2. Setting the discoverability to 1 in blueutil as a regular user and checking on the same devices.
3. Setting the discoverability to 1 in blueutil as a root user and checking on the same devices.

In an attempt to remedy this issue I have:
- Restarted the Bluetooth module.
- Restarted the machine.
- Updated the machine to a new MacOS minor version.

I suppose a last ditch effort could be to reinstall MacOS, however, I'm a little loathe to do this since it's a bit of a minor issue.

I've also tested running the above command on another MacBook Pro with similar results occurring. I'll list the specs of the two laptops below, for some extra info (this could be a Catalina specific issue, maybe?)
- Macbook Pro 16 (2019)
- Macbook Pro 15 (2015)

On a side note, both systems above were able to use blueutil as a regular user (before encountering the bug) and set discoverability such that they were able to show up on other devices.

## Extra Information

Things also get a little weirder if you check the discoverable values as a regular user and a root user. The two values don't seem to coincide? Eg: If discoverable is 0 for both root/a regular user and I set discoverable as root, the regular user will still see discoverable as zero. For a further example:

```bash
blueutil -d
> 0
sudo blueutil -d
> 0
sudo blueutil -d 1
sudo blueutil -d
> 1
blueutil -d
> 0
```

My guess is that something has been written to that shouldn't have been written to as root? Honestly, this is a little outside my expertise. I'm a bit of a layman when it comes to the inner-workings of MacOS. Any input on a solution to this would be much appreciated!

Brikwerk, 2020-10-30:
I've had some time to experiment with this issue more.

Unfortunately, things have gotten a little weirder. I restarted my main machine once more for other reasons and now I actually can't get the system to *stop* being discoverable. This means I have the exact opposite issue from what I wrote above. I ran through the above steps I detailed to try and correct, yet still I can't toggle discoverability.

Anyways, since I'm functionally in the same position (unable to control discoverability), I went ahead with creating a new user, testing discoverability, and looking at the Bluetooth plist. I'll detail some of the things I tried below in chronological order.

- After creating the new user, rebooting my machine, and logging into the new account, I double checked and the system was still stuck as discoverable.
- I opened the bluetooth preferences menu and closed it. After this, the system was still stuck as discoverable.
- I tried to set discoverability on and then off as a regular user (no sudo) with blueutil. This was also unsuccessful at changing the discoverable state.
- Finally, I tried to set discoverability as root with blueutil. This also was unable to change the state.

I did notice, however, that non-root blueutil writes to `~/Library/Preferences/ByHost/com.apple.Bluetooth.*.plist` just fine. I was able to observe this value toggling on and off with the appropriate commands. I also noticed that root blueutil doesn't write to this value, however, changes with root blueutil are still tracked. It seems to be writing elsewhere, possibly.

toy, 2020-10-31:
I'm already thinking that to prevent the problem code can be added that will check process uid not to be 0 and adding an environment variable to override the restriction, but better to find how to fix it first to ensure that it is really caused by setting discoverability as root.

Few things you can try:
- Maybe worth checking if there is some weird file created in the system library preferences path: `/Library/Preferences/ByHost/com.apple.Bluetooth.*.plist` or by running `find /Library/Preferences -iname '*bluetooth*'`
- Resetting bluetooth module, which will be available in bluetooth menu under Debug if you open the menu while holding ⌥⇧
- Resetting SMC is often mentioned as a possible solution
- Trying a bluetooth USB dongle if you can get a hold of one

dave4jr, 2020-11-14:
@toy @Brikwerk Hey guys, I've also ran into this issue doing the same thing as above. My bluetooth can no longer connect to any device. I've noticed that when opening Bluetooth Explorer > Tools > HCI Controller Selector, there's no longer any controller in the list here. I actually did try reinstalling the entire operating system and it did not help. I just updated to BigSur and still an issue. I've purchased another USB Bluetooth and that is working. Also, in my console, I'm seeing an error happen over and over:
[image]

Brikwerk, 2020-11-14:
@dave4jr I think our machines are exhibiting slightly different issues. I'm able to pair/connect/reconnect with other Bluetooth devices on the affected machines. The only symptom I'm seeing is the inability to toggle discoverability. Also, I believe the HCI Controller Selector won't actually list the default Apple Bluetooth adapter. I quickly checked this on an unaffected Mac and "None" was listed.

@toy I've found time to do some more debugging with the steps you listed in your reply.
- I ran the above command and manually searched the directories you mentioned. I wasn't able to find any extra or odd Bluetooth related files, apart from the regular `com.apple.Bluetooth.plist`.
- Resetting the Bluetooth module has no affect
- Resetting the SMC seems to have kicked my system from being "always discoverable" to "never discoverable". This could also be caused by the simple act of restarting since this was originally how my system ended up as "always discoverable".
- Using a USB Bluetooth dongle has no affect

I do, however, have some other interesting results:
- The `com.apple.Bluetooth.plist` file (located at /Library/Preferences/com.apple.Bluetooth.plist) seems to reflect the current discoverable state of my system. The current state also seems to be accurately reflected by holding the ⌥ key, clicking on the Bluetooth menu bar icon, and reading the discoverable value. This value does not change if the Bluetooth preferences pane is opened or normal/root blueutil is run.
- The Bluetooth plist is readable/writable by the system only. I'm not 100% if this is how it should be or if my user should also be included. I've tried adding my user to this list as read/write and running through toggling discoverability. Afterwards,I still wasn't able to change the discoverability of the system.
- Upon opening the Bluetooth preferences pane, the "Date Modified" for the Bluetooth plist is immediately updated to the current time, so it would seem that the preferences pane is attempting to make writes?
- I'm seeing consistent error messages in the Console when I open the Preference pane:
[image]

- The `mdwrite(665)` error only occurs when I open the Bluetooth preferences pane. The `Bluetoothd(208)` errors will occur slightly after the preference pane is opened. I think these errors might be a first clue to what's going wrong, possibly?

dave4jr, 2020-11-14:
@Brikwerk Oh okay, thanks for the clarification. I'm a software developer but all the iOS dev I've done is through React Native so I'm really lacking in knowledge when it comes to diagnosing this issue, and the fact that my bluetooth seems completely bricked is really unfortunate. I know it was after I ran this program. I tried all of the solutions you mentioned in your original post as well. Have you seen the console error message I posted in my last message? "CoreBluetooth isn't advertising on client type 9". I have no idea what that is and have tried searching high and low on the web. Any help would be truly appreciated!

toy, 2020-11-14:
@dave4jr Sounds weird. If you have reinstalled the system, then the problem is not in system files/configs as it most probably is for @Brikwerk. Can you describe exact steps that led to the problem that you have? It is probably better to extract it as a separate issue.

@Brikwerk I've successfully reproduced the issue. After running `sudo blueutil -d 1` discoverability state got stuck in discoverable state. I used `fs_usage -w` in parallel and saw operations by `cfprefsd` with `com.apple.Bluetooth.*.plist` under `/private/var/root/Library/Preferences/ByHost` (directory are not readable, so `sudo find /private/var/root/Library/Preferences/ByHost -name 'com.apple.Bluetooth*'` can be used) and `/Library/Preferences/com.apple.Bluetooth.plist`.
What I found after lots of restarts, retries, deleting files and trying to use `defaults`:
- A key `DiscoverableState` with right value is added to `/private/var/root/Library/Preferences/ByHost/com.apple.Bluetooth.XXX.plist` (where XXX is machine identifier) when using `sudo blueutil -d X`
- `DiscoverableState` is also added to `/Library/Preferences/com.apple.Bluetooth.plist` but not right away (apparently can be triggered by switch discoverability without sudo) and value in it is really showing the discoverable state, but it is not changing with the value in the other file
- Using `defaults` to try to delete the key didn't help
- But deleting both files and restarting afterwards fixed the problem:
  ```
  sudo find /private/var/root/Library/Preferences/ByHost -name 'com.apple.Bluetooth*' -print -delete
  sudo rm /Library/Preferences/com.apple.Bluetooth.plist
  ```
So unfortunately the only solution that worked for me involves resetting bluetooth preferences. If it is fine for you could you please try it? Otherwise solution is to somehow delete the `DiscoverableState` key.

toy, 2020-11-14:
@Brikwerk It would be great if you could try https://github.com/toy/blueutil/compare/refuse-to-run-as-root

Brikwerk, 2020-11-14:
@toy Thanks for looking into this more and biting the bullet by reproducing! I can also confirm that deleting the two files you mentioned and restarting the affected systems was able to fix both.

I also built/ran the new branch and I can also confirm that the preventative measures are effective. I'm able to toggle discoverability normally as a non-root user. Running as root and attempting to toggle prints the new message and prevents triggering the bug.

toy, 2020-11-17:
Awesome!

I've merged the change and released https://github.com/toy/blueutil/releases/tag/v2.7.0.
I assume this issue can be closed and @dave4jr can open a new issue if needed.

Brikwerk, 2020-11-17:
Great to hear! Thanks for your help/work on this issue @toy! Very much appreciated. 
I also think with the solution implemented, we can close this issue. 

@dave4jr If your Bluetooth problems persist, I would open up another issue since your problem does sound like it might be more hardware-centric vs. configuration-based.

closed 2020-11-17 by Brikwerk