toy/blueutil #42: Any support for macOS 10.13.6?

thevickypedia, 2020-11-28:
I'm using High Sierra and `brew install blueutil` keeps throwing 
`Executable requires at least macOS 10.14, but is being run on macOS 10.13.6, and so is exiting.`

Any solution would be much appreciated. Thanks in advance.

toy, 2020-11-28:
To me this looks like a problem with `brew` executable or some other tool it is using under the hood to compile the binary as it apparently doesn't have it compiled for version before 10.4.

There should be nothing preventing `blueutil` from running even on 10.9 (e74824825c89dc5a1c665a448ef46d3e2d9d12b5). Minimum version is specified both for building using [make](https://github.com/toy/blueutil/blob/master/Makefile#L1) and [xcodebuild](https://github.com/toy/blueutil/blob/master/blueutil.xcodeproj/project.pbxproj#L216) (which is [apparently used by homebrew](https://github.com/Homebrew/homebrew-core/blob/master/Formula/blueutil.rb#L20)), so binaries should be able to work on all version starting with 10.9.

Easiest solution would be to [build from source](https://github.com/toy/blueutil#from-source).

thevickypedia, 2020-11-28:
> To me this looks like a problem with `brew` executable or some other tool it is using under the hood to compile the binary as it apparently doesn't have it compiled for version before 10.4.
> 
> There should be nothing preventing `blueutil` from running even on 10.9 ([e748248](https://github.com/toy/blueutil/commit/e74824825c89dc5a1c665a448ef46d3e2d9d12b5)). Minimum version is specified both for building using [make](https://github.com/toy/blueutil/blob/master/Makefile#L1) and [xcodebuild](https://github.com/toy/blueutil/blob/master/blueutil.xcodeproj/project.pbxproj#L216) (which is [apparently used by homebrew](https://github.com/Homebrew/homebrew-core/blob/master/Formula/blueutil.rb#L20)), so binaries should be able to work on all version starting with 10.9.
> 
> Easiest solution would be to [build from source](https://github.com/toy/blueutil#from-source).

I thought the error message from `blueutil` as I didn't read the entire console output. My bad.
Thanks for the solution 🙂

closed 2020-11-28 by thevickypedia