toy/blueutil #43: Inquiry failing

johnathanchiu, 2021-01-01:
I currently have one MacBook Pro (13-inch, 2018) running Big Sur v11.1 and when I call ```blueutil --inquiry``` and attempt to find nearby devices it often fails and does not work consistently. On the other hand, I have a second MacBook Pro (13-inch, 2015) running Mojave v10.14.6 which does the exact same functions and always works each time. Is there any logical explanation for this? Maybe updates to the framework or could it just be an issue with my bluetooth hardware/software. If so, is there any way to reset the bluetooth module?

toy, 2021-01-02:
Does it fail on the 2015 MacBook with those devices for which it fails on the 2018 one?

Also does system preferences Bluetooth pane work correctly?

Resetting bluetooth module and other things that can help from another issue:
- Resetting bluetooth module, which will be available in bluetooth menu under Debug if you open the menu while holding ⌥⇧
- Resetting SMC is often mentioned as a possible solution
- Trying a bluetooth USB dongle if you can get a hold of one

You can try checking if it is hardware/software related if you have a spare hdd/sdd by installing there the other OS version, so trying booting 10.14 on 2018 model or 11.1 on 2015.

toy, 2021-03-21:
@johnathanchiu Please reopen if needed

closed 2021-03-21 by toy