toy/blueutil #44: Bad CPU type when using Automator Run Shell Script

kravetsya, 2021-01-25:
I have Macbook M1 2020 with Big Sur 11.1
I've installed HomeBrew and blueutil in order to create keyboard shortcut for BT switch on/off
Blueutil works fine through Terminal, but don't when creating Run Shell Script in Automator

Script is following

export PATH=/usr/local/bin:$PATH
bluetoothStatus=$(blueutil -p)
[[ $bluetoothStatus -eq 0 ]] && blueutil -p 1 || blueutil -p 0

On run i have mistake 
The action “Run Shell Script” encountered an error: “-: line 1: /usr/local/bin/blueutil: Bad CPU type in executable
-: line 2: /usr/local/bin/blueutil: Bad CPU type in executable
-: line 2: /usr/local/bin/blueutil: Bad CPU type in executable”

toy, 2021-02-03:
The project is distributed only as source files and the best solution is for Homebrew to provide universal binaries or platform specific binaries.
You can check the type of binary by running `file /usr/local/bin/blueutil`, if it is not arm64 it will probably mention only `x86_64`.

Your script can be much simpler as you can call `blueutil -p toggle` after setting environment. There are instructions for your goal in the wiki: https://github.com/toy/blueutil/wiki#instuctions-to-create-a-keyboard-shortcut-that-toggles-bluetooth.

The error looks weird because there should be no such difference between running script from Terminal and from Automator. Unless somehow there is difference in the environment and you have multiple `blueutil` binaries in different paths and in one case first one is i386 (probably better to run `which -a blueutil` in Terminal and Automator to check).
I also assume that the problem is not `blueutil` specific, so using any binary with same architecture should cause this error.
But as I don't have any M1 computer, I can't test the problem myself.

kravetsya, 2021-02-10:
Thank you so much

"Unless somehow there is difference in the environment and you have multiple blueutil binaries in different paths and in one case first one is i386 (probably better to run which -a blueutil in Terminal and Automator to check)."

you're absolutely right. 
which -a blueutil  gives the following

/opt/homebrew/bin/blueutil
/usr/local/bin/blueutil

i've changed script as following

export PATH=/opt/homebrew/opt/blueutil/bin:$PATH
bluetoothStatus=$(blueutil -p)
[[ $bluetoothStatus -eq 0 ]] && blueutil -p 1 || blueutil -p 0

everything works fine now, and keyboard shortcut also works ok.

kravetsya, 2021-02-10:
Thank you so much

"Unless somehow there is difference in the environment and you have multiple blueutil binaries in different paths and in one case first one is i386 (probably better to run which -a blueutil in Terminal and Automator to check)."

you're absolutely right. 
which -a blueutil  gives the following

/opt/homebrew/bin/blueutil
/usr/local/bin/blueutil

i've changed script as following

export PATH=/opt/homebrew/opt/blueutil/bin:$PATH
bluetoothStatus=$(blueutil -p)
[[ $bluetoothStatus -eq 0 ]] && blueutil -p 1 || blueutil -p 0

everything works fine now, and keyboard shortcut also works ok.

by the way, i have 2 installations: from original source and from homebrew

kravetsya, 2021-02-10:
Finally i used wiki and create the following script

/opt/homebrew/opt/blueutil/bin/blueutil -p toggle

toy, 2021-02-10:
> the issue comes from i've installed first from http://www.frederikseiffert.de/blueutil/
> since original isn't adopted to M1, i've installed it from homebrew.

Ironic is that the binary you've installed from there is also universal with two architectures, but both are not supported already before Big Sur: `ppc_7400` and `i386`.
 
> is it necessary to uninstall first one? can you help with the command to do it?

It is probably a good idea as the binary will not work and can be confusing again at some point. If you've used the `Install.command`, then the only thing to uninstall is to remove the file `/usr/local/bin/blueutil`, so `rm /usr/local/bin/blueutil` maybe prefixed with `sudo`.

kravetsya, 2021-02-11:
Thanks a lot.
Issue is resolved

closed 2021-02-11 by kravetsya