toy/blueutil #46: Workflow to reconnect a keyboard to another mac

JanC, 2021-02-12:
Hey @toy 
many thanks for this tool and for having it open source 👍 

I'm often switching a keyboard from on mac to another. My current workflow that _somehow_ works without using `blueutil` is to

- on mac1, remove completely the keyboard in the BT preferences
- on mac2, wait until the keyboard appears in the device list (turn BT on/off a couple of times)
- on mac2, click on `Connect`

I'm looking for a way to simplify it and found  your tool. However, it seems I have a similar issue to this one https://github.com/toy/blueutil/issues/38

I'm aware the removing/unmpairing completely the device does not seem to be possible as explained in https://github.com/toy/blueutil/issues/31

I try by using a simple disconnect as follows

**on mac1**
```
blueutil --info $my_keyboard
address: 11-22-33-44-55-66, connected (master, -47 dBm), not favourite, paired, name: "MyKeyboard", recent access date: 2021-02-12 08:51:45 +0000

blueutil --disconnect $my_keyboard
```

**on mac2**

```
blueutil --info $my_keyboard
address: 11-22-33-44-55-66, not connected, not favourite, paired, name: "MyKeyboard", recent access date: 2021-02-12 09:03:56 +0000

blueutil --connect $my_keyboard
blueutil --info $my_keyboard

address: 11-22-33-44-55-66, not connected, not favourite, not paired, name: "MyKeyboard", recent access date: 2021-02-12 09:07:16 +0000

blueutil --info $my_keyboard
address: 11-22-33-44-55-66, connected (master, -50 dBm), not favourite, not paired, name: "MyKeyboard", recent access date: 2021-02-12 09:07:46 +0000

blueutil --pair $my_keyboard

blueutil --info $my_keyboard
address: 11-22-33-44-55-66, not connected, not favourite, paired, name: "MyKeyboard", recent access date: 2021-02-12 09:08:31 +0000

blueutil --connect $my_keyboard
blueutil --info $my_keyboard
address: 11-22-33-44-55-66, not connected, not favourite, not paired, name: "MyKeyboard", recent access date: 2021-02-12 09:08:45 +0000

```

It basically seems it's impossible to pair or connect they keyboard on the mac2 unless I manually unpair un from mac1. 

Are there any other commands to try to be able to switch a keyboard from one mac to another somewhat easily? Would you consider adding the unapair command that would use [this api](https://stackoverflow.com/questions/14964561/is-it-possible-to-unpair-a-bluetooth-device-in-cocoa-objc/20749542#20749542) even if it needs a root access?

toy, 2021-02-13:
Hey Jan
First thank you for easily understandable issue and for checking previous ones :)

Short version - did you try (temporarily) turning bluetooth off on mac1?

As I understand bluetooth devices will normally try to connect to last connected device.
And I've just checked if I can come up with decently reliable way to switch a keyboard between two macs (I'll call them mac1 and mac2 too). Both are paired with the keyboard and have bluetooth turned on.

If I rotate bluetooth power on mac1 using: `blueutil -p 0; sleep 5; blueutil -p 1` and press some key on the keyboard, then not very reliably but it connects to mac2, a bit weird is that it seems to be doing it after sleep on mac1 finishes (or it already turns power back on), so decreasing sleep time makes connection happen faster, but also decreases reliability.
Then I checked if just starting `blueutil --connect "xxx"` on mac2 and right after that running `blueutil --disconnect "xxx"` on mac1 will be enough, but it didn't work for me.
What finally worked well was to start `blueutil --connect "xxx"` on mac2 and right after that running `blueutil -p 0; sleep 5; blueutil -p 1` on mac1. Switching commands I was able to jump between computers pretty quickly.
Is it an option for you and if yes please check if it works?

About unpairing api - I'm fine to try it, but check if the other solution works for you.

JanC, 2021-02-15:
turning the BT off on mac1 (even completely) does not seem to help. I noticed that on mac2, after I issue the `connect` command, the keyboard is in a kind of limo state where it's `non paired` but `connected`

```
blueutil --pair $my_keyboard
Failed to pair "xyz" with error 0x05 (Authentication Failure)

blueutil --info $my_keyboard
connected (master, -41 dBm), not favourite, not paired, name: "MyKeyboard", recent access date: 2021-02-15 09:25:35 +0000

blueutil --info $my_keyboard
not connected, not favourite, not paired, name: "MyKeyboard", recent access date: 2021-02-15 09:25:35 +0000
```

In the System Preferences, it appears in the device list without the "Connect" button but also without a subtitle that indicates whether it's connected or not.

[image]

It all works flawlessly when I manually unpair the keyboard  on mac1 and then on mac2:

```
blueutil --pair $my_keyboard
blueutil --connect $my_keyboard
blueutil --info $my_keyboard
address: xyz, connected (master, -41 dBm), not favourite, paired, name: "MyKeyboard", recent access date: 2021-02-15 09:34:10 +0000
```

I understand that this is more a Bluetooth / macOS question so feel free to close this ticket.

toy, 2021-02-16:
I noticed in your original post that paired state was jumping which looks weird. Maybe it is worth trying to reset bluetooth module and/or erasing paired devices at least on mac2. One way of doing it is from Debug menu in bluetooth menu item if you open it while pressing ⌥⇧.

toy, 2021-05-04:
After having no luck at all with the method described on stackoverflow:
* tried using `SCPreferencesCreate` but no unpairing was happening and plist was shortly overwritten
* trying `SCPreferencesCreateWithAuthorization` I understood that the `SCPreferences` interface is really not intended for anything but network configuration
* tried `CFPreferences` which is the right interface for modifying preferences, but like `SCPreferencesCreate` it doesn't unpair

I then stumbled upon an undeclared method 🎉
It will be great if you can try the changes in branch [unpair](https://github.com/toy/blueutil/compare/unpair)

JanC, 2021-05-05:
hey,
thanks for looking into this. 

it seems to work nicely:

on mac1:

```
./blueutil --unpair $my_keyboard
```

on mac2:

```
./blueutil --pair $my_keyboard
./blueutil --connect $my_keyboard
```

👍

toy, 2021-05-05:
> I'd suggested wrapping [this line ](https://github.com/toy/blueutil/compare/unpair#diff-c81affcfada2cbf544a72c7135060ddedf09d05d94a2913b9059561b3ad5c00dR1025)in a `if respondsToSelector` to avoid crashes

Thanks!

closed 2021-05-23 by toy

toy, 2021-05-23:
Just released v2.9.0 which includes unpairing.

wds15, 2022-01-22:
Just found this and started using it...works like a charm. Thanks!

BhavanaPrabhakar, 2022-08-01:
i'm trying to pair iPhone device with Mac mini by using following command

blueutil --pair xx-xx-xx-xx-xx-xx

but i'm getting - this error 

Does "iPhone-QA1" (xx-81-xx-27-xx-66) display number 384659 (yes/no)? YES
Failed to pair "xx-81-xx-27-xx-66" with error 0x05 (Authentication Failure)

is there anyway, i can remotely connect bluetooth between iphone and mac mini?

toy, 2022-08-01:
@BhavanaPrabhakar Is failure appearing instant after confirming or after some time?
I think iPhone may also be asking for confirmation even though it didn't initiate pairing.

BhavanaPrabhakar, 2022-08-02:
> 

it appears instantly. i tried with yes and no, multiple times but, i get the same error all the time. 

i usually connect to iPhone remotely using Vysor. But at first, it needs to be connected to Bluetooth and this is one time activity.  But due to system restart or something bluetooth connection is disconnected and i'm trying to connect using blueutil via terminal

toy, 2022-08-02:
@BhavanaPrabhakar Do you need to re-pair iPhone or re-connect? If it is already paired then `blueutil --connect xx-xx-xx-xx-xx-xx`

BhavanaPrabhakar, 2022-08-02:
currently it says "connection failed"(screenshot below) 

[image]

I'm trying to pair and connect but pair is giving me this error - 

Does "iPhone-QA1" (xx-81-xx-27-xx-66) display number 384659 (yes/no)? YES
Failed to pair "xx-81-xx-27-xx-66" with error 0x05 (Authentication Failure)

toy, 2022-08-03:
The screenshot seems to suggest that the device is already paired, is it not?

commit 889250eb (2021-05-05): add unpairing functionality, resolves #27, resolves #31, resolves #32, resolves #46, resolves #53
commit 1a5a5c61 (2021-05-05): add unpairing functionality, resolves #27, resolves #31, resolves #32, resolves #46, resolves #53

Kudos to @JanC for suggesting how to avoid crash if the undeclared
method that is used is not available.
commit 2921fa98 (2021-05-05): add unpairing functionality, resolves #27, resolves #31, resolves #32, resolves #46, resolves #53

Kudos to @JanC for suggesting how to avoid crash if the undeclared
method that is used is not available.
commit 05ce3862 (2021-05-23): add unpairing functionality, resolves #27, resolves #31, resolves #32, resolves #46

Kudos to @JanC for suggesting how to avoid crash if the undeclared
method that is used is not available.