toy/blueutil #48: Device ID name sometimes not found

trwarren, 2021-02-15:
When I use the -info command giving a device name that is returned from the --paired command I get back an error message "Device not found by name"; however if I give the address string then it is successful. 

The device name does have a # character in it (device name is "Magic Mouse #1") and it seems that this is what causes the problem. If I access another device by name, "Magic Mouse 2" that works ok. So:

./blueutil --info "Magic Mouse #1" fails
./blueutil -- info "Magic Mouse 2" succeeds

toy, 2021-02-16:
I tried to reproduce it, but adding " #1" suffix to bluetooth device name didn't prevent me from getting info by name. My first guess would be to check that there are no extra/hidden/weird characters in the name, maybe by renaming it or checking the byte sequence that is returned using `--paired`.

trwarren, 2021-02-17:
thanks for the response; I did rename the device (still has #) and it worked ok; so I guess the original name must have had some unicode character (or whatever).

closed 2021-02-17 by trwarren