toy/blueutil #51: Wait for *multiple* devices to connect

Matsuo3rd, 2021-03-25:
@toy: thanks for this great tool and making it available to the community.

I know --wait-connect ID is EXPERIMENTAL, but would you consider enriching the feature with waiting for *multiple* devices to connect.

My workflow is the following: I can connect multiple Sony Dualshock game controller to my Mac. Once *any* of them is connected, I am starting some script. I was hoping to give device name to --wait-connect so it would be triggered by any controller (they are all named "DUALSHOCK 4 Wireless Controller". But it looks like current behaviour is to resolve device name with first device ID in the recent list. Meaning, only one of the controller will trigger the --wait-connect (first in the list).

So, would it be possible, with an extra flag, to wait for *any* devices matching a device name pattern?

`
blueutil -v
2.8.0

blueutil --recent
address: 1c-a0-b8-a5-4c-35, not connected, not favourite, paired, name: "DUALSHOCK 4 Wireless Controller", recent access date: 2021-03-25 08:15:07 +0000
address: 90-89-5f-0a-f2-0f, not connected, not favourite, paired, name: "DUALSHOCK 4 Wireless Controller", recent access date: 2021-03-25 08:14:47 +0000
address: 90-89-5f-f1-a5-ba, not connected, not favourite, paired, name: "DUALSHOCK 4 Wireless Controller", recent access date: 2021-02-24 20:30:59 +0000

blueutil --wait-connect "DUALSHOCK 4 Wireless Controller"
`
==> Will only be triggered when "1c-a0-b8-a5-4c-35" is connected, not the other ones.

toy, 2021-03-28:
Thank you for opening the issue, looks interesting.
Just to clarify, I marked some functionality experimental more to say that the usage is not part of stable api and can change, rather than that it can be removed.

Currently all actions requiring device id when called with name will resolve it to latest device with matching name in the list of recently used, so depending on order of connecting or disconnecting your controllers the resolved one should change.
Your case is when multiple devices have same name and this can be handled by replacing device resolution by name, with checking connected device to have the right name.

I came up with following options:
* one more parameter for `--wait-*` commands to wait for any device by name,
* some specifier for name like `any:DUALSHOCK 4 Wireless Controller` (but `any:` can be a prefix for device name too),
* some switch (like `--match-all-by-name` or something) to tell all commands to work with any device by name.

I'm just not sure yet how to make it not confusing itself and compared to other commands (like for example `--is-connected` could be also checking for any or all devices with specified name to be connected instead of checking the resolved one).
May there be a need to wait for all devices with some name, and maybe waiting for any of differently named devices (like either keyboard or mouse)?
`--wait-connect`, `--wait-disconnect`, `--wait-rssi` and `--is-connected` will probably benefit from all possibilities.
For `--info`, `--add-favourite`, `--remove-favourite`, `--connect` and `--disconnect` an option for name to resolve to all devices with that name could be good.
`--pair` is probably better left with specifying one device, but while investigating I understood that for a device unknown to system it is not possible to pair with device by its name even after inquiry.

Matsuo3rd, 2021-03-29:
Thanks for considering my ask. I would go with option 3 (--match-all-by-name).
It is more clean and flexible.