toy/blueutil #52: Will this be developed to include services & attributes?

stormbytes, 2021-04-02:
Have been looking high and low for a command line utility or Python library that works on OSX. Need to scan adverts, show device info along with services and UUIDs, etc. Both LightBlue and LightSee do this in GUI but I need a CLI. Was hoping this would be it. Alas, no. If anyone has any tips, please comment!

toy, 2021-04-02:
Could you please explicitly describe the desired functionality?

stormbytes, 2021-04-02:
It’s described pretty clearly in the project description. What specifically
are you asking?

stormbytes, 2021-04-02:
Apologies, I confused two different messages! The ability to scan
discovery, get services, characteristics, read/write. That sort of thing.

toy, 2021-04-02:
I assume by "scan discovery" you mean scanning for nearby devices (already present as `blueutil --inquiry`), "get services" seems clear (no such functionality yet in the tool), but is "characteristics, read/write" related to those services?

stormbytes, 2021-04-03:
Ideally the tool should pretty much function like BlueSee. Allowing
discovery of nearby peripherals, querying those peripherals to obtain their
services information and characteristics. The tool should also allow for
read/write characteristics and subscribing to notification. I've spent a
good week looking around and so far I've found exactly nothing that simply,
cleanly and elegantly accomplishes this for BLE on Mac OSX. Blueutil is the
closest ideological candidate so far.

toy, 2021-04-03:
I finally understood that you need this functionality for BLE, but `blueutil` is currently using only IOBluetooth framework which provides access to "classic" Bluetooth devices and not CoreBluetooth which provides access to BLE.

stormbytes, 2021-04-03:
Ah, well.. that's that. Any plans to support BLE in future versions?

toy, 2021-04-03:
I see nothing against including support for BLE, but didn't yet plan it too. Probably better to rename the issue to describe missing function and mention BLE.