toy/blueutil #53: Magic Mouse 2 doesn't connect

kassi, 2021-04-09:
Hi Ivan, nice to meet you here :-)

after switching from Magic Mouse 1 to MM 2, my auto connect doesn't work anymore.
I'm using 2 Macbooks and usually my bluetooth mouse follows my cable keyboard. For that I'm doing

    mac1: blueutil --disconnect <id> && blueutil -p 0
    mac2: blueutil --connect <id>

which worked fine in both directions even without holding mouse button down.
With magic mouse 2 the `--connect` doesn't do anything anymore (using Catalina 10.15.7, blueutil 2.8.0).
Device is listed as paired but not connected.
If I open bluetooth settings and hit the connect button next to the mouse, it connects.

Any idea?

toy, 2021-04-11:
Hi Karsten, nice to meet you here too, funny every time worlds intersect :)

Commands are really what I would suggest with just two notes: I would start command on mac2 before starting commands on mac1, AFAIR it worked more consistently; also not sure if disconnect is really needed before switching bluetooth off, but I doubt it can harm either.

Normally for me if there is a problem connecting, it is not working both using `blueutil` and GUI.
Is it not connecting to both macs that way?

I suggest to first check if `blueutil --info <id>` gives you expected result.
If yes the simplest thing to try is to unpair and pair again.

toy, 2021-05-06:
I stumbled upon an undeclared way to unpair devices, details in [comment for #46](https://github.com/toy/blueutil/issues/46#issuecomment-832326520).
Would be great if you can confirm that changes in branch [unpair](https://github.com/toy/blueutil/compare/unpair) work for you too.

kassi, 2021-05-21:
Hi @toy - sorry for the late reply. Finally found some time.
Today I tested the new unpair functionality. At first: unpairing works on both systems with  my MM2 (High Sierra and Catalina) without problems.

However it doesn't help in any regards with my problem. If I unpair, I can't simply pair again because I'm getting "Authentication Failure" most of the time.
I even came to a state now (on Catalina), where hitting the connect button in the system preferences yields an error (though not telling me what happened, just that it was unsuccessful).
I think the problem with the MM2 is that, in order to pair, you have to switch it off and on.

To my problem above:
I pair it on both devices, connect to Catalina and disconnect again.
As soon as I connect on the High Sierra machine, although it's paired, it doesn't connect. It appears in the sys pref with a connect button at first, but after some seconds the whole entry gets lost and `bllueutil --paired` doesn't show the MM2 anymore. Repairing needed, which is always a hassle.
If I swicth the mouse off and on when swicthing devices, it starts connection (without being unpaired), but it always shows a dialog box with authentication where I have to hit one button. This is odd. Any hint how I can get around this?

toy, 2021-05-23:
First thanks for confirming that unpairing works!
Sad that it doesn't help you though. Some time ago I had problems with reconnection of devices including through preferences pane and menulet, and only one of removing `/Library/Preferences/com.apple.Bluetooth.plist` before restart and resetting PRAM and resetting SMC helped me.

> I think the problem with the MM2 is that, in order to pair, you have to switch it off and on.
I assume devices when they loose connection should switch to a mode in which they at least should accept connection attempts, but still "turning it off and on again" can correct something in internal state.

If it shows authentication dialog and it is paired, then something should be wrong with its state. I suggest trying resetting the mouse, and bluetooth module and if nothing helps removing bluetooth plist before restart and/or resetting PRAM/SMC.

commit 889250eb (2021-05-05): add unpairing functionality, resolves #27, resolves #31, resolves #32, resolves #46, resolves #53
commit 1a5a5c61 (2021-05-05): add unpairing functionality, resolves #27, resolves #31, resolves #32, resolves #46, resolves #53

Kudos to @JanC for suggesting how to avoid crash if the undeclared
method that is used is not available.
commit 2921fa98 (2021-05-05): add unpairing functionality, resolves #27, resolves #31, resolves #32, resolves #46, resolves #53

Kudos to @JanC for suggesting how to avoid crash if the undeclared
method that is used is not available.