toy/blueutil #55: Pairing Issues on BigSur (previously working on Mojave)

jwignall, 2021-06-29:
Hey there, 

Firstly thank you for making and sharing this tool, I've found it very useful. 

I had been using blueutil on Mojave with no problems to pair and unpair devices as needed. However, I just upgraded to BigSur and I'm having issues. Now when I try to pair to device, it no longer connects to it as part of the pairing process. If I run --connect <mac> after pairing, it will say it's connected, but I won't see the Bluetooth device in Audio MIDI Setup (it's an audio device), and the device's Bluetooth indicator will not suggest it is connected. e.g.: 
1) blueutil --pair <mac>   // pairs device, but device will appear in preferences as "not connected" - --info says the same
2) blueutil --connect <mac> // it will say it is connected, but it isn't showing as an audio device
So right now, I am unable to get a pairing to work via this tool. 

What is strange is that if I do the initial pairing manually, it works; I can pair and unpair multiple times. Eventually (not sure what causes it, I think restarting the computer), it will stop working again and I'll need to repair manually. It's all very odd but this never happened on Mojave...

Any help would be much appreciated. 

I'm using v2.9.0

toy, 2021-07-03:
Firstly thank you :D

So if I understand correctly pairing using `blueutil` somehow does the process incomplete. Did you also try disconnecting and connecting back after initial pairing (instead of unpairing/pairing)? And do you have any other similar devices to try?
For now I can't yet help with BigSur.

jwignall, 2021-07-06:
Hi there, 

Yeah it seems that way; my expectations from my experiences on Mojave is that `--pair` should take the device from unpaired to paired and connected, whereas I am just getting paired but not connected, and trying to use `--connect` doesn't properly complete the process either as my bluetooth device will not show up in Audio MIDI Setup. 

I have tried disconnecting and connecting; it works as long as the initial pairing was carried out via the UI and not blueutil.  I have tried on two devices and seem the same results each time. 

Thanks!

joaoasilva, 2021-08-12:
Hi all,

I'm having the same issues. It connects fine when I pair but after a restart/going to sleep it displays as connected but doesn't work. The only way to fix it again is going to the OS Bluetooth GUI, unpair and pair again.
I'm using a bluetooth keyboard (Logitech G613) and a bluetooth mouse (Logitech G603).

Thanks in advance.

gazsiazasz, 2021-09-21:
I have an issue on Monterey with pairing the A1016 Apple Wireless Keyboard.
With running the `./blueutil --pair 00:0A:95:CC:BB:AA 000000 ` it gives the following output:

```
Input pin 000000 on "Gábor billentyűzete"
2021-09-21 21:38:19.062 blueutil[2351:47063] IOBluetoothDevicePair BluetoothHCIPINCodeRequestReply - 0
2021-09-21 21:38:19.656 blueutil[2351:47066] -[IOBluetoothDevicePair peerPairingCompleted:withError:]: Calling connectionComplete on IOBluetoothDevicePair. Status: 31
Failed to pair "00:0A:95:CC:BB:AA" with error 0x1f (Unspecified Error)
```

On the PacketLogger.app it looks like that the Keyboard requests (the keyboard sends the initial PIN code request) a 6 digit PIN, but the computer is replying with a 4 digit PIN (all 0s).