toy/blueutil #60: `--inquiry` not working, but Bluetooth Preferences MacOS GUI correctly discovering mouse - Catalina & Big Sur

cristiano-belloni, 2021-11-04:
I'm trying to pair one of my computer with my Logitech MX Master 2S.
- If I set the mouse on pairing mode and run `blueutil --inquiry` repeatedly nothing is found (command returns 0, "success").
- If I go to my Bluetooth Preferences GUI, instead, the mouse is correctly discovered (see images below)

[image]

[image]

This happens with blueutil 2.9.0, installed via brew on Catalina (MacBook Pro (Retina, 15-inch, Mid 2015)) and Big Sur (MacBook Pro (16-inch, 2019).

CHBrianLiu, 2021-11-14:
Same issue here. I couldn't find the Bluetooth device from the `--inquiry` command, but the device was visible from the Preference UI.

[image]

toy, 2021-11-14:
Thank you both for reporting, do you have possibility to check same actions with different bluetooth device and to check same device with different computer?

CHBrianLiu, 2021-11-15:
Hi @toy, thanks for checking this. I think I can help test on another device.

I've tried on another Mac, MacBook Pro (13-inch, M1, 2020), which runs MacOS 11.4. My Magic Keyboard (not the same keyboard I reported above, sorry that I don't have it with me now) can be detected by `blueutil`. And all the peripherals presented in the System Preferences are in the list as well.

If there's anything that can help you troubleshoot, I'm willing to provide more information or logs. Thanks for the kind help.

toy, 2021-11-16:
@CHBrianLiu Thank you for checking, this means that it at least doesn't affect all systems running 11.4.

Would be great if you can also check inquiry with some other device on affected system and if the issue stays after system restart.

fedekrum, 2021-12-01:
Using Catalina. It may be something related to the more restricted access of apps and scripts to BT and other security things that have to be approved at the "Security and Privacy" Control panel.

fedekrum, 2021-12-01:
Yesssss !!! That is !!!
You must go to "System Preferences - Security & Privacy - Privacy (tab) - Developer Tools" and allow "Terminal" application to run software without a system security policy.

Is not a bug, is a matter of permission.

fedekrum, 2021-12-01:
[image]

toy, 2021-12-03:
@fedekrum Thanks for investigating! My attempt at reproducing was not completely successful - I wanted to check if allowing bluetooth interaction  (instead of "Developer Tools") will also work, but after it started working with the option it didn't stop working after disabling and removing the option, and even after restarting.

@cristiano-belloni @CHBrianLiu Can you try checking if permissions are the issue?

RaphaelKn, 2022-03-27:
Hi @toy 
There seem to be a bug in the --inquiry on Monterey (at least on 12.3) where in many cases the inquiry doesn't return for a long time.
I managed to fix it - can i post a side branch and open a PR (i don't seem to have permissions for that..)?
Thanks

siutsin, 2023-01-12:
Hi @toy I have a similar issue with my Logitech MX Anywhere 3 mouse, but NO problem with the Logitech M557 mouse, Apple Magic Mouse, and Keychron K10 keyboard.

I have to add iTerm to **BOTH**  Developer Tools and Bluetooth in System Preferences - Security & Privacy - Privacy to see the Logitech MX Anywhere 3 mouse with `blueutil --paired` - adding only either of these will not be visible.

I am using macOS 12.6.2 and am unsure what the difference is between the new mouse and other devices, but it is problematic as I don't have admin right on some of the laptops I would like to pair. I am wondering if there is any way to pair it up without the admin privilege?

toy, 2023-01-14:
@siutsin Checking if I can reproduce I noticed that without adding Bluetooth permission for terminal app (or by turning it off) I don't see my connected mouse both using `blueutil --paired` and `blueutil --connected`, but permission doesn't affect seeing a headset and a keyboard. Also no Developer Tools permission for the terminal. So looks like it is down to understanding the difference between devices from os perspective.

fbukevin, 2023-06-22:
Same issue to me

milindprajapati, 2024-04-08:
I am seeing an issue with blueutil --inquiry on macOS Ventura. Is anyone facing same issue? It will get stuck  at blueutil[900:16498] -[IOBluetoothDeviceInquiry initWithDelegate:] -  0x600002568090

toy, 2024-04-11:
@milindprajapati Can you have a look if #66 works for you?

ankou29666, 2024-06-10:
> I am seeing an issue with blueutil --inquiry on macOS Ventura. Is anyone facing same issue? It will get stuck at blueutil[900:16498] -[IOBluetoothDeviceInquiry initWithDelegate:] - 0x600002568090

same issue for me on 2015 mbp on Sonoma with OCLP. Getting value 0x600003fec020 instead. 

keyboard (logitech craft) paired successfully via preferences on Monterey. Sonoma discovers the keyboard but never prompts for PIN code. But did prompt me for PIN when pairing my phone. MX master 3S paired smoothly also.

toy, 2024-06-10:
@ankou29666 I assume OCLP stands for [OpenCore-Legacy-Patcher](https://dortania.github.io/OpenCore-Legacy-Patcher/), if yes than it may be one more reason for framework or api problems on top of existing issues with supported hardware/os.
The `0x600003fec020` is most probably just memory address, the line is logged by framework and I didn't yet find a way for it to not log the line.
I also don't really understand the description, do you have inquiry or pairing problem?
Overall are you using the latest version of blueutil, as it contains an attempt to fix the inquiry problem?

ankou29666, 2024-06-10:
the last version being supported on 2015 macbook pros being Monterey, OpenCore Legacy Patcher is the only way to have a more recent version installed.

toy, 2024-06-11:
> the install was made around two monthes ago

@ankou29666 Please check the version then, as v2.10.0 was released less than 2 months ago

ankou29666, 2024-06-12:
I confirm it's 2.10.0

mykmelez, 2025-01-15:
I just tried `blueutil --inquiry`, with both 2.10.0 from Homebrew and a local build on the tip of the main branch, on my MacBook Pro M1 running Sonoma 14.7.2, and I don't see any results, regardless of whether or not I specify a timeout:

> \> blueutil --inquiry
> 2025-01-15 10:47:48.985 blueutil[79292:1919346] -[IOBluetoothDeviceInquiry initWithDelegate:] -  0x600002f94030
> 2025-01-15 10:47:59.481 blueutil[79292:1919346] -[IOBluetoothDeviceInquiry dealloc] -  0x600002f94030