toy/blueutil #62: Can't connect/disconnect by device name on Monterey

malcolmsailor, 2021-12-09:
Hi, since upgrading to Monterey, I can no longer use device names (e.g., 'STR-DH190') with commands like `blueutil --connect` or `blueutil --disconnect`. This isn't a huge problem since addresses like `3c-91-80-1a-e6-ba` still work.

FWIW I also have the same issue with `--disconnect` mentioned in #58 ; either of the workarounds (adding `--info` or `--wait-disconnect`) mentioned in that thread work.

deeeeekun, 2024-01-27:
Having the same issue on Sonoma

toy, 2024-01-27:
@deeeeekun If you list recent devices (`blueutil --recent`), do you see the name that you use?

deeeeekun, 2024-01-29:
@toy No, it doesn't show either

toy, 2024-01-29:
You mean the list is empty or there are no names?
What about `blueutil --paired`?
For now only recent connected devices are used for device names, so if the list is empty or no names, blueutil will not be able to have devices referenced by name.

deeeeekun, 2024-02-02:
The device (and device name) shows up in `blueutil --paired`. `blueutil --recent` returns an empty list, even though I have just connected to it (using the address).

toy, 2024-02-03:
Looks like a good idea to make that blueutil first checks paired devices for names. Though absence of recent devices must be a bug, framework, or os installation/state issue.

PR #88 (2024-04-17): Also search paired devices to get device by name
IOBluetoothDevice.recentDevices return empty list in macOS Monterey and
later, also search paired devices to make

  --info "BY NAME"
  --connect "BY NAME"
  --disconnect "BY NAME"
  ...

work again.

Fix #62
commit af643417 (2024-04-17): Also search paired devices to get device by name

IOBluetoothDevice.recentDevices return empty list in macOS Monterey and
later, also search paired devices to make

  --info "BY NAME"
  --connect "BY NAME"
  --disconnect "BY NAME"
  ...

work again.

Fix #62
closed 2024-04-17 by toy