toy/blueutil #63: --recent and --favourites not working on macOS Monterey

friedrichweise, 2021-12-11:
I'm the developer of [wechsel](https://github.com/friedrichweise/wechsel/) and experiencing some issues in macOS Monterey. I'm using the last Intel Macbook from 2015, that actually supports macOS 12.
Running the following `blueutil`-command always returns an empty list:

```
$ blueutil --recent
$ blueutil --favourites
```

At wechsel I'm using the following Swift code to access the devices. Using `recentDevices()` or `favoriteDevices()` – either way the device list is empty:

```
guard var devices = IOBluetoothDevice.recentDevices(0) as? [IOBluetoothDevice] else {
            print("Error accessing IOBluetoothDevice.recentDevices")
            return []
}
```

There seems to be a problem with the [IOBluetooth](https://developer.apple.com/documentation/iobluetooth) framework on Monterey. Anyone experiencing similar issues?

toy, 2021-12-11:
Simplest way to check is by using a script in swift:
```swift
#!/usr/bin/env swift

import IOBluetooth

print(IOBluetoothDevice.favoriteDevices()!)
print(IOBluetoothDevice.recentDevices(3)!)
```

Can you check if it is related to permissions as in #60?

friedrichweise, 2021-12-12:
The scripts returns two empty arrays:
```
$ ./test.swift
[]
[]
```

I've read Issue https://github.com/toy/blueutil/issues/60 and added the Terminal to the "Developer Tools" privacy section. This doesn’t affect the behaviour. Is there an elegant way to get the full system log using the Console.app or a similar tool?

toy, 2021-12-12:
Can you try adding the Bluetooth privacy setting (and restart Terminal)?

For system log there is `log` command, but it may be tricky to filter for such case. For example everything on all levels containing case insensitive blue: `log stream --level debug --color always --timeout 10s | grep -i blue`.

friedrichweise, 2021-12-12:
Alright. I added the Terminal (and also iTerm2 – my primary terminal) to the Bluetooth and the Developer Tools privacy settings. Then run `blueutil --favorites`. Given your `log` command I get the following output:

```
blueutil: (libsystem_info.dylib) Retrieve User by ID
bluetoothd: [com.apple.bluetooth:Server.Core] PowerManager power state is 1
bluetoothd: [com.apple.bluetooth:Server.Core] PowerManager power state is 1
bluetoothd: [com.apple.bluetooth:Server.A2DP] No corresponding device for DF963AC9 found in fA2dpHandleToDeviceMap (0 devices)
bluetoothd: [com.apple.bluetooth:Server.A2DP] isStreamingAudioToDevice: Unable to retrieve handle for device.
bluetoothd: [com.apple.bluetooth:Server.A2DP] No corresponding device for 994ABEEE found in fA2dpHandleToDeviceMap (0 devices)
bluetoothd: [com.apple.bluetooth:Server.A2DP] isStreamingAudioToDevice: Unable to retrieve handle for device.
bluetoothd: [com.apple.bluetooth:Server.MacCoex] Posting Bluetooth Status Notification <private>
studentd: (ClassroomKit) [com.apple.classroom:Bluetooth] entered finishedRescanInterval
PerfPowerServices: (CoreBluetooth) [com.apple.bluetooth:CoreBluetooth] -[CBClassicManager retrievePairedPeersWithOptions:] <private>
PerfPowerServices: (CoreBluetooth) [com.apple.bluetooth:CoreBluetooth] -[CBClassicManager retrievePairedPeersWithOptions:] <private>
PerfPowerServices: (CoreBluetooth) [com.apple.bluetooth:CoreBluetooth] Sending synchronous XPC message <private>: <private>
bluetoothd: [com.apple.bluetooth:Server.XPC] Received XPC message "CBMsgIdRetrievePairedPeersWithOptions" from session "com.apple.PerfPowerServices-classic-164-17"
bluetoothd: (CoreFoundation) [com.apple.defaults:User Defaults] looked up value <private> for key [SOME MAC] in CFPrefsPlistSource<0x7fb00020d020> (Domain: com.apple.MobileBluetooth.devices, User: kCFPreferencesCurrentUser, ByHost: No, Container: /Library/Bluetooth, Contents Need Refresh: No) via CFPrefsSearchListSource<0x7fb000115150> (Domain: com.apple.MobileBluetooth.devices, Container: /Library/Bluetooth)
bluetoothd: [com.apple.bluetooth:Server.Core] TagSet's class is not an NSArray, returning error for  69422B50
bluetoothd: [com.apple.bluetooth:Server.Security] 69422B50 hasTS: 0, res: 1
bluetoothd: (CoreFoundation) [com.apple.defaults:User Defaults] looked up value <private> for key [SOME MAC] in CFPrefsPlistSource<0x7fb00020d020> (Domain: com.apple.MobileBluetooth.devices, User: kCFPreferencesCurrentUser, ByHost: No, Container: /Library/Bluetooth, Contents Need Refresh: No) via CFPrefsSearchListSource<0x7fb000115150> (Domain: com.apple.MobileBluetooth.devices, Container: /Library/Bluetooth)
bluetoothd: [com.apple.bluetooth:Server.Core] TagSet's class is not an NSArray, returning error for  683AA17B
…
bluetoothd: [com.apple.bluetooth:Server.Security] 9C6B4312 hasTS: 0, res: 1
bluetoothd: [com.apple.bluetooth:Server.LE] Already tracking address "<private>" as device "[SOME MAC]"
bluetoothd: [com.apple.bluetooth:Server.Core] Get SDP record data from SDP attributes for device 69422B50
bluetoothd: [com.apple.bluetooth:Server.Core] SDP record cache size 1200 with 12 services
bluetoothd: [com.apple.bluetooth:Server.Core] Returning sdp record NSData: <private>
bluetoothd: [com.apple.bluetooth:Server.XPC] bluetoothd sending sdp data: <private>
bluetoothd: [com.apple.bluetooth:Server.LE] Already tracking address "<private>" as device "[SOME MAC]"
bluetoothd: [com.apple.bluetooth:Server.Core] Get SDP record data from SDP attributes for device 683AA17B
bluetoothd: [com.apple.bluetooth:Server.Core] sdp attributes is empty

```

fsonntag, 2022-01-04:
@toy Any updates on that? :)

toy, 2022-01-04:
@fsonntag Sadly no, checking on the surface showed that methods are broken and I didn't yet have time to create environment to investigate deeply. For now I can suggest checking if `blueutil --paired` or `blueutil --connected` can be of help.

friedrichweise, 2022-01-05:
`blueutil --paired` and `blueutil --connected` work like a charm…

@fsonntag are you able to reproduce the bug? My speculation is that this bug only affects Intel based devices that run macOS 12.

fsonntag, 2022-01-16:
@friedrichweise 
Sorry for the late reply, somehow missed this :/ 

Anyway, both commands run without an issue, no error output.

wwwjfy, 2022-01-30:
`[IOBluetoothDevice recentDevices]` returns empty. I made the following change to make connecting by name work, in function `get_device`. could be a macOS bug (Intel only? I'm using Mac mini 2018 and MBP 2019). #58 #62 should be of the same cause.

```
diff --git blueutil.m blueutil.m
index 47aa91d..b5225d5 100644
--- blueutil.m
+++ blueutil.m
@@ -277,7 +277,7 @@ IOBluetoothDevice *get_device(char *id) {
       exit(EXIT_FAILURE);
     }
   } else {
-    NSArray *recentDevices = [IOBluetoothDevice recentDevices:0];
+    NSArray *recentDevices = [IOBluetoothDevice pairedDevices];

     if (!recentDevices) {
       eprintf("No recent devices to search for: %s\n", id);
```

toy, 2024-04-10:
I can only document that functionality is broken, concerning fixing finding devices by name, see #88

commit 30fbc488 (2024-04-10): document macOS >= 12 silently ignoring favourites and recent access date

Resolves #63
Resolves #84
closed 2024-04-10 by toy