toy/blueutil #64: --connect for device is not holding through blueutil, but connection can hold through the Mac GUI

jdonahue58, 2022-01-10:
using blueutil v2.9.0 on Mac Big Sur 11.6, connect is working very temporarily. Having the macOS GUI up, I can see the device connects using blueutil --connect through a script I wrote and manually, and then a few seconds later shows "not connected." I thought it might be the device, but when I connect through the GUI, the connection holds.

jdonahue58, 2022-01-10:
[blueutil_connect.mov.zip](https://github.com/toy/blueutil/files/7840827/blueutil_connect.mov.zip)

jdonahue58, 2022-01-10:
after more digging I believe this has something to do with the pairing. The --connect feature works with devices that do not require pin pairing. I see someone reported a similar issue #55

Also, is there a feature to turn off the automatic prompt for pin pairing? I would like to be able to use this feature in an automated fashion

toy, 2022-01-13:
Thanks for opening the issue.

Could you please tell if you did intentionally upload the .zip of a .mov and not .mov itself? It will be easier to understand the issue if a player is presented instead of a link.

From the recording you are connecting a device that is already paired, so pin should not be an issue unless you are unpairing every time.

Also if you mean bypassing pin when pairing it could decrease security of pairing. As long as you understand the consequences in case during pairing only confirmation is requested, you can pipe `yes` like `yes | blueutil --pair …`.

jdonahue58, 2022-01-14:
Hi, yes the file was too big to upload the video by itself. 

I only see this issue with devices that require pin pairing. It looks as though it pairs fine and shows up in blueutil --paired:
```
jdonahue$ blueutil --paired
address: c4-3a-35-dd-54-cc, not connected, not favourite, paired, name: "Mustang GTX", recent access date: 2022-01-14 16:25:47 +0000
```

When I try to connect after that is when I see the issue:
```
jdonahue$ blueutil --connect "Mustang GTX"
Failed to connect "Mustang GTX"
```

toy, 2022-01-14:
On video the command succeeds, but connection drops right after, in your last comment the command failed.

Just to check if you tried cycling bluetooth power and restarting?

jdonahue58, 2022-01-14:
I just tried it and same result:
```
usdjdonahue:~ jdonahue$ blueutil --power off
usdjdonahue:~ jdonahue$ blueutil --power on
usdjdonahue:~ jdonahue$ blueutil --inquiry
address: c4-3a-35-dd-54-cc, not connected, not favourite, not paired, name: "Mustang GTX", recent access date: 2022-01-14 16:25:47 +0000
usdjdonahue:~ jdonahue$ yes yes | blueutil --pair "Mustang GTX"
Does "Mustang GTX" (c4-3a-35-dd-54-cc) display number 633681 (yes/no)? usdjdonahue:~ jdonahue$ blueutil --connect "Mustang GTX"
Failed to connect "Mustang GTX"
```
I believe connection drops because it tries to connect during the pairing command. Attached is a video that shows that. I've experienced both scenarios.

jdonahue58, 2022-01-14:
Sorry, my mistake. I did not mean to close the issue. 

I just tried it and same result:
```
usdjdonahue:~ jdonahue$ blueutil --power off
usdjdonahue:~ jdonahue$ blueutil --power on
usdjdonahue:~ jdonahue$ blueutil --inquiry
address: c4-3a-35-dd-54-cc, not connected, not favourite, not paired, name: "Mustang GTX", recent access date: 2022-01-14 16:25:47 +0000
usdjdonahue:~ jdonahue$ yes yes | blueutil --pair "Mustang GTX"
Does "Mustang GTX" (c4-3a-35-dd-54-cc) display number 633681 (yes/no)? usdjdonahue:~ jdonahue$ blueutil --connect "Mustang GTX"
Failed to connect "Mustang GTX"
```
I believe connection drops because it tries to connect during the pairing command. Attached is a link for a video that shows that. I've experienced both scenarios. 1. Where it attempts to connect during the pairing, does not hold, and then it "connects" using the command but then it drops connection immediately after. 2. I use the command and I get the message "Failed to connect." 

[(https://youtu.be/a85QKyjyLKM)](url)

toy, 2022-01-14:
Odd is that on the video in the GUI the device after pairing appears connected, disconnects, connects again and disconnects, then connects after `blueutil --connect` (and disconnects again).
Please try also `blueutil --connect "Mustang GTX" --wait-disconnect "Mustang GTX"`.

If paired using `blueutil`, will it connect properly through menu item? And vice versa?

jdonahue58, 2022-01-14:
Seems like I can pair through blueutil, connect through the Mac GUI fine and connection holds. There is no option in the GUI to just pair, so I connect through the GUI, then disconnect and it is still in the list of paired items in blueutil. Then try to connect through blueutil and get "failed to connect."
```
usdjdonahue:~ jdonahue$ blueutil --inquiry
address: c4-3a-35-dd-54-cc, not connected, not favourite, not paired, name: "Mustang GTX", recent access date: 2022-01-14 23:28:32 +0000
usdjdonahue:~ jdonahue$ blueutil --disconnect "Mustang GTX"
usdjdonahue:~ jdonahue$ blueutil --paired 
address: c4-3a-35-dd-54-cc, not connected, not favourite, paired, name: "Mustang GTX", recent access date: 2022-01-14 23:31:26 +0000
usdjdonahue:~ jdonahue$ blueutil --connect "Mustang GTX"
Failed to connect "Mustang GTX"
```
I still am seeing failure using --wait-disconnect as well

jdonahue58, 2022-01-18:
Here's a video of my process. For some reason the first run of `--connect "Mustang GTX" --wait-disconnect "Mustang GTX"` didn't work but I was able to run it again and it worked. I also wanted to show the connection through the GUI working. Let me know if you'd like to see anything else. 

https://youtu.be/JYOixsKjlX4

toy, 2022-02-17:
Hi Jillian,
Unfortunately no updates.
Good thing is that the problem may correct itself with the system upgrade. Please write on the outcome if you go with the upgrade.

fallible-algebra, 2024-01-26:
Hi, I just wanted to contribute to this issue as I had been struggling with it this morning, also on macos and with both --connect and --wait-connect (both terminated immediately while in a script).

I've had a different but similar issue to this and a few other issues being tracked on github. Strangely, they were resolved by running:

`bash -c "blueutil --connect ID"`

instead of `blueutil --connect ID`. I'm bringing this up not as a pertinent detail but instead as a fix for anyone else who is coming across this issue.

toy, 2024-01-27:
@fallible-algebra That is peculiar, this creates one more process between terminal emulator and blueutil, but there is already the login shell, so no idea why would it fix the issue.
Is bash also your login shell?
I assume it should also work, but can you try also zsh or some scripting language (`ruby -e 'system "blueutil --is-connected ID"'`)?

fallible-algebra, 2024-01-28:
Zsh is my login shell, with a fairly basic oh-my-zsh install.

fallible-algebra, 2024-01-29:
Can confirm that specifically `ruby -e "system 'blueutil --connect ID'"` also successfully connects, and that the command you posted does correctly report the connection status.