toy/blueutil #65: will this allow RFCOMM like functionality on osx12

SpudGunMan, 2022-01-31:
question will this project allow SDP/RFCOMM (serial port functionality) on OSX 12.1?

it seems that there was a change on the new osx12 been having issues finding a way around that.. will this project allow proper use of the OSX tty ports or able to shim back in?

toy, 2022-01-31:
I'm not certain I understand the questions.
Currently there is no functionality related to SDP or RFCOMM in the tool and I can imagine adding service discovery, but not sure how to fit RFCOMM.

SpudGunMan, 2022-01-31:
understood thank you, also looks like osx12.3 has some fixes to serial issues

brandonros, 2024-03-20:
how do you achieve SDP/RFCOMM via Bluetooth on MacOS?

toy, 2024-03-21:
IOBluetooth framework has multiple classes related to SDP and RFCOMM, look for `IOBluetoothSDP` and `IOBluetoothRFCOMM`