toy/blueutil #67: Inquiry hanging in macOS 12.4

rfardy, 2022-07-14:
Intel Mac running macOS 12.4 and blueutil 2.9.0

When I run --inquiry, it hangs about 90% of the time. When it does work, it typically takes between 40 and 65 seconds. I've tried running it as a child process with timeouts of 10, 30, 60, 90 and 120 seconds, without success.

I've also tried all the recommended Bluetooth resets.

However, the following works consistently on 12.4 for both Intel and M1.

In one terminal window, I run blueutil --inquiry for about 10 seconds.
In a second terminal window, I run blueutil --power 0 (after inquiry has been running for 8-10 seconds).
As soon as Bluetooth power is disabled, blueutil immediately reports back with a list of unpaired, visible devices.

On my Intel and M1 /Library/Preferences/com.apple.Bluetooth.plist never gets modified. If I delete it, it is not recreated after a Bluetooth reset.

/Library/Bluetooth/Library/Preferences/com.apple.MobileBluetooth.devices.plist seems to be the only plist getting modified, along with the sqlite3 databases in the /Library/Bluetooth directory.

Any ideas? Thanks in advance!

toy, 2022-07-17:
Thank you for detailed report. Feels like it may have common underlying problem with few other issues (#58, #64)

Noah40, 2023-10-24:
Im on Sonoma 14.0 with Intel as well and the same thing is happening to me :(