toy/blueutil #70: Disconnect/wait-disconnect issues

Kosai1, 2022-10-15:
Hello, 
I am having issues when trying to --disconnect or --wait-disconnect. 
The command runs but the bluetooth device doesn't disconnect. When I use --wait-disconnect, i get an error that states "Timed out waiting for "(null)" to disconnect" even though I properly inputed the right bluetooth MAC address. 
Thanks for creating the tool!

toy, 2022-10-16:
Few questions:
Does this device have a name (checking through preference pane or using `bluetil --connected`/`blueutil --paired`/`blueutil --recent`)?
Could you try disconnecting another device same way?

Kosai1, 2022-10-16:
Hello, thank you for your quick response. 
Yes the devices do. 
Yes, I have tried it with multiple devices. The first two are my google nest speakers. The second was my bluetooth headphones. 

All connect but don't disconnect via terminal/blueutil. I can disconnect them via the menu bar or by using apps like ToothFairy. 

If you need more info, please let me know!

hb1998, 2022-11-12:
I have the exact same issue, it connects but doesn't disconnect. 
FYI: I have the same issue with [bluetoothconnector](https://github.com/lapfelix/BluetoothConnector)
I'm using a TWS (One plus Buds Z)

paulmiddelkoop, 2022-11-23:
@JoshuaIyalagha You say that you try `--disconnect` or `--wait-disconnect`, but I found out that you need to combine them.

This works for me:
```
blueutil --disconnect 04-fe-a1-e4-73-75 --wait-disconnect 04-fe-a1-e4-73-75
```

I debugged a bit because I tried to fix a bug in the bluetooth extension (Toothpick) for Raycast. 
I noticed that the underlying problem is that `IOBluetoothDevice.closeConnection` returns successfully  but does not close anything. A simple delay between fetching the device and calling `closeConnection` seems to fix the problem. I did not fount an elegant solution yet to fix it in Raycast.

See also https://github.com/lapfelix/BluetoothConnector/issues/20 where people experience the same problems.

paulmiddelkoop, 2022-11-23:
@toy Maybe this gives you some information to fix this problem on the blueutil side?

toy, 2022-11-26:
@paulmiddelkoop I'll write thoughts around the problem.

Documentation of `closeConnection` says `This method is synchronous and will not return until the connection has been closed (or the command failed). In the future this API will be changed to allow asynchronous operation.`, so it doesn't behave according to documentation. I don't know if it is overall a bug somewhere in framework or has something to do with the fact that it is called from a CLI binary (some assumption on part of framework making it not work well in this case). May be clarified by creating a simple GUI app with a button to call `closeConnection` on a device.

Unless the reason for the problem is discovered, I imagine 2 solutions (besides relying on always combining with `--wait-disconnect`):
* Create new command, something like `--disconnect-and-wait`, downside is creating a command only to overcome the bug, which behaves same if the bug gets fixed or if it is not affecting certain version/environment.
* Change `--disconnect` to additionally wait for disconnection, seems that it should not change the behaviour, but maybe I'm missing a case where it does, then it may be introducing unintended behaviour.

paulmiddelkoop, 2022-11-27:
@toy I would go with option 2. I think most users expect that a -`-disconnect` just does what is says. I would not expect that people rely on an async disconnect now, but if there do they rely on a ondocumented behaviour of blueutil.

Also I think it's best that the `--connect' and `--disconnect` behave the same way. So either both being async or both being sync.  Since the `--connect` is synchronous, it makes sense to me to do the same thing for `--disconnect.`

toy, 2023-01-02:
Would be great if someone having the problem can check if [changes on wait-on-disconnect branch](https://github.com/toy/blueutil/pull/new/wait-on-disconnect) fix them

paulmiddelkoop, 2023-01-10:
@toy I tested the branch and it fixes it 🚀  Great job!

closed 2023-01-14 by toy

toy, 2023-01-14:
[v2.9.1 is out](https://github.com/toy/blueutil/releases/tag/v2.9.1)

@JoshuaIyalagha Could you please check if the issue is fixed for you?

Tocseoj, 2023-03-29:
Just adding that I was having the same issue with my magic mouse, but upgrading to v2.9.1 fixed it. Thanks, @toy!

commit d2c1452c (2023-01-14): wait for disconnect when requesting disconnect, fixes #70
commit c9c3c15c (2023-01-14): wait for disconnect when requesting disconnect, fixes #70