toy/blueutil #71: A1016 Keyboard and Ventura

jwrw, 2023-01-27:
I've a M1 Mac with 13.1 Ventura.  I originally used blueutil to get my nice old A1016 keyboard connected.  However, something changed and I ended up losing connectivity, deleted the pairing but couldn't get it back.  Judging by comments on various forums, there's been a lot of work around the BT stack that has been breaking things.  On the latest O/S, blueutil itself does seem to behave slightly differently than I might have expected.  (e.g. why do I have to turn off BlueTooth to get the scan results to appear?)

Back to the A1016...  I'm not sure if this counts as an issue but I wanted to share something that worked.  I think the keyboard sends a PIN when you connect, which may conflict with the newer BT standards.  If you just do `blueutil --pair <kdb Mac> 0000` then switch on the keyboard and hit `<return>` on the A1016, you get an authentication failure. 

**BUT** If you immediately repeat the `blueutil --pair <kdb Mac> 0000` command and type `0000<return>` on the A1016 keyboard, pairing is successful.  You can then connect the keyboard using the BT panel in System Settings.

[NB.You will clearly need a connected keyboard to enter the blueutil command]

toy, 2023-01-27:
Normally it is expected for devices that have a screen to present the pin for confirmation and for devices that can input numbers to input the same pin. So `blueutil --pair ID PIN` expects for pairing with a keyboard the PIN to be typed on the keyboard. Not sure if PIN can be only numbers on any characters, but not reqired to be 0000. You then should be able to also use `blueutil --connect ID` to connect without opening System Setting/System Preferences.

jwrw, 2023-01-28:
Agree - the behaviour is 'abnormal' - it's quite an old keyboard.  Other keys are ignored when you switch it on but the first press of `return` initiates a pairing with key `0000` from the keyboard. This fails, because blueutil like most hosts expects to initiate pairing with a keyboard.  I guess any PIN would do in the command - as the first one is ignored when pairing fails and you can enter the second one on the A1016.  Using the standard Apple Bluetooth pairing, MacOS throws up a PIN code in a pop-up when you initiate the connection with the <return> key.  However, this immediately then disappears, as the keyboard sends its '0000' PIN and authentication fails.  So blueutil has certainly dug me out of a hole and allowed be to get paired! Thx!

vlastoun, 2023-02-28:
I have the same problem with Logitech mouse MX3 and keyboard Logitech Mechanical. I am not able to --inquiry or --pair them ( they can be paired over macos bluetooth ui). I found that its address is changed after every successfull paired atempt. Last two digits are increasing. But when running inquiry it does not work.. If i guess next correct adress then I try to pair and no success. I also try to put pin there and no success. 

Is there any way I can debug it? or more verbose?

toy, 2023-03-05:
@vlastoun One of the first links in search suggests that Logitech devices can do something odd about their mac address https://unix.stackexchange.com/q/690808. Problem with pairing is maybe the yet unresolved issue on newer osx versions.

a2mta, 2023-06-24:
Hello @jwrw. 
I wanted to ask if you were able to do that trick once or if it's something that works all the time. 
It's just I bought the same keyboard the other day and was disappointed by the lack of macos support and don't want to throw it in the closet because of that. 
I have the same issue you described with almost the same setup (M1 mac pro with macos Ventura 13.4.1), and I could not reproduce your solution. 
blueutil just fails with two different errors ("Unspecified Error" and "No connection")
Maybe it has something to do with the timings when you have to press "return" or something else. ?

vlastoun, 2023-06-28:
No, I have never resolved my issue. It is happening for all my Logitech devices 24. 6.

tamruta, 2023-10-04:
I was able to connect my A1061 with M2 Mac Pro 2022 using `blueutil --pair <kdb Mac> 0000`

I've done this a few times now and my pairing has been stable. The kb connects automatically to my mac once I have paired

toy, 2023-10-05:
@tamruta Great to hear that it works for you. Do you mean A1016? Also see [my comment](#issuecomment-1407092010) about the pin.

jwrw, 2023-10-05:
I think I managed the trick with the A1016 a couple of times - but once I got it paired I didn't want to risk trying it over and over. It is quite likely that you need to be a little lucky with the timing?!?!

tamruta, 2023-10-06:
@toy Sorry I meant A1016 

When my kb is low on battery it disconnects more easily. If I don't type something as soon as it connects, the connection is lost. In that situation I just keep an eye on the green light by the power switch to connect. 

I think I needed more luck to get the mac address of the kb than to pair with it. I was using `--inquiry` and it wouldn't show up unless the kb was switched as soon as I had ran `--inquiry`.

toy, 2023-10-06:
@tamruta Unless you unpair the keyboard, you shouldn't need to pair it to connect, so `--connect NAME` should be enough, also you can use name and not the address for that.

tamruta, 2023-10-06:
@toy Ah sorry for the confusion, I was just mentioning that for notes. I've not had any trouble connecting once it has been paired. Thanks for the tip! I haven't tried using the name

strangerbit, 2023-10-21:
> Hello @jwrw. I wanted to ask if you were able to do that trick once or if it's something that works all the time. It's just I bought the same keyboard the other day and was disappointed by the lack [...]

I have exactly the same issue.
> blueutil --pair 00-01-A4-xx-xx-xx 0000
Failed to pair "00-01-A4-xx-xx-xx" with error 0x02 (No Connection)

jwrw, 2023-10-26:
### Config
> Apple Studion M1 Max, Sonoma 14.0 (recently upgraded from Ventura but steps are the same)<br>
> blueutil 2.9.1
> Apple Wireless Keyboard (aka A1016) 1.1.0

(My A1016 was connected and working fine so I went to bluetooth settings, clicked the little 'i' next to the keyboard and selected 'Forget Device...' and clicked the scary red 'Forget Device' on the warning dialog.)

### Preamble
Plug in a generic USB keyboard - all typing is on that keyboard unless specified.  If you're doing this on a laptop, then you don't need the USB keyboard, the input will be on that laptop keyboard.

### Get the keyboard mac address
| Terminal | Terminal 2 | A1016 | Note |
| - | - | - | - |
| Open a Terminal window | Open a second terminal window |  A1016 switched off |
| Enter `blueutil --inquiry` | | | Get 1 line confirmation inquiry is running |
| | | Switch on keyboard |
| | After 10s enter `blueutil -p 0` | | Power off bluetooth |
| *Results of inquiry now appear* | | |
| | Enter `blueutil -p 1` | | Put bluetooth back on |
| *Find mac address of keyboard here* | You can close Terminal 2 now | Switch off keyboard | |
| | | | |

### Pair the A1016
You're using the USB (or laptop) keyboard to enter Terminal commands and the A1016 is switched off.
| Terminal | A1016 | Note |
| - | - | - |
| Type `blueutil --pair 00-0a-95-xx-xx-xx 0000` BUT DON'T HIT 'Enter' YET | | |
| | Swtch on keyboard | | |
| Hit 'Enter' | | Within a few seconds |
| Get message *Input pin 0000 on "Apple Wireless Keyboard" (00-0a-95-xx-xx-xx)* | | |
| | Hit 'Enter' | |
| Get message *Failed to pair "00-0a-95-xx-xx-xx" with error 0x1f (Unspecified Error)* | |
| Fairly quickly press 'Up Arrow' then 'Enter' to repeat the `blueutil --pair 00-0a-95-xx-xx-xx 0000` command | | This is just what I did - I don't know how fast you need to be but a few seconds seems OK |
| Get message *Input pin 0000 on "Apple Wireless Keyboard" (00-0a-95-xx-xx-xx)* | | |
| | Type `0000` and press 'Enter' | Send the PIN from the A1016, again fairly quickly after entering the `pair` command |

If you have the bluetooth settings open you should see the keyboard appear as connected. Hit 'Enter' and it should all be OK.

adamesq, 2023-10-27:
Unfortunately, I cannot get this to work on M2 Mac Mini running Ventura...I've tried over 20 times. I've got the MAC address using the method above (and double-checked with a mid-2014 MBP), but almost always gives a no connection error on the M2. _Occasionally,_ I get Unspecified Errors, but NEVER get prompted for the Input pin. Just a big fail on this end.

toy, 2023-10-29:
@jwrw About «Get the keyboard mac address» - inquiry gets stuck or just takes much more time?
About «Pair the A1016» - does it not work if typing `0000` and pressing 'Enter' on first attempt?

@adamesq Do you have bluetooth permission on for your terminal? See https://github.com/toy/blueutil/issues/82#issuecomment-1772548214

jwrw, 2023-10-31:
Hi @toy, I tried leaving the command for a good 10 minutes and got no response, other than the initial confirmation
```
~ % blueutil --inquiry
2023-10-31 09:17:28.028 blueutil[81517:1660588] -[IOBluetoothDeviceInquiry initWithDelegate:] -  0x6000026bc080
```

If I leave the command for a little time (e.g. 1 minute) then try the `blueutil -p off ; blueutil -p on` trick, I get to see only the BT devices that are not connected (i.e. I don't see the BT keyboard and mouse that are connected)

In this case the output is like...
```
~ % blueutil --inquiry
2023-10-31 09:32:36.621 blueutil[83381:1678866] -[IOBluetoothDeviceInquiry initWithDelegate:] -  0x6000001c8050
address: 90-0f-0c-xx-xx-xx, not connected, not favourite, not paired, name: "the big TV", recent access date: 2023-10-31 09:32:58 +0000
address: 00-1d-df-xx-xx-xx, not connected, not favourite, not paired, name: "SoundSticks Wireless", recent access date: 2023-10-31 09:32:58 +0000
2023-10-31 09:32:58.125 blueutil[83381:1678866] -[IOBluetoothDeviceInquiry dealloc] -  0x6000001c8050
```

jwrw, 2023-10-31:
Re: pairing - I tried all sorts of combinations, certainly 0000<enter> was the first thing, but the keyboard BT doesn't seem to be active until that first <enter>.  Doing it a second time immediately after, as per the above,  was what worked for me.

lnealan, 2023-11-03:
> Thank you for the suggestion. I had not, but after adding the permission, I now get Connection Errors. In both cases (before and after enabling, I never get to the request to enter 0000. 🤔
> 
> EDIT: After posting the reply above, I tried just hitting "enter" after inputting the pair request. Between the connection errors, I started seeing the request to enter 0000. After about 15 randomly timed requests, I hit "0000 " while waiting for a connection error. It just happened to be timed correctly. 'Input pin 0000 on "00:0x:xx:xx:xx:xx"' just happened to show up while I was inputing and...now connected. Kinda hit and miss. I appreciate the help here, but not the hoops we are all having to jump through in lieu of a simple fix in the Mac OS!

I had the same issue of never getting the Input pin prompt and was only able to finally get it after resetting the PRAM on my macbook pro. Once that was working i was able to successfully pair as well.

hou-mon, 2023-11-07:
I too had the same issues of never getting the input pin prompt. A solution that's worked for me and seems reproducible was to add the device as a favorite BEFORE attempting to pair. The keyboard doesn't need to be on to do this:

`blueutil --add-favourite 00-0a-95-xx-xx-xx`

From that point I would follow @jwrw pair instructions.

newusersm, 2023-11-21:
Hello! I have read a lot of pages to connect the A1016, but I still can't do it (MacBook Pro 16 M1 Pro) I have installed blueutil but it doesn't work for me either. It never asks me to enter 0000 code but I do it anyway and I do get the error Ox1f, 0x02 and sometimes 0x04. Any solution? I really like the keyboard and I would like to use it.

Note: I have realized that on Mac OS Sonoma it is impossible for me to correctly execute blueutil --inquiry and blueutil --discoverable 1 . Is this normal or am I doing something else? Poor solution of deleting the two files that you recommended on another page but it didn't work.

[image]

paulzaptec, 2023-11-29:
I had a similar problem with only ever getting "No Connection" or "Unspecified Error" on macOS Sonoma / MBP M1. This seems to be because I had the Bluetooth settings window open.

Closing the Bluetooth settings window allowed me to successfully pair an A1016 keyboard with: 
`blueutil --pair 00-0a-95-xx-xx-xx 0000 `

dariolob, 2023-12-06:
Hey there. I've tried every possibile combination but I'm still unable to pair my A1016 to Sonoma 14.1 on Mac Mini M1. I followed @jwrw instructions religiously, bluetooth permissions are in place, I've tried with the BT settings window both open and closed, also tried adding the device as a favorite like suggested by @hou-mon... nothing. Do I have to give up? :(

alexcu, 2024-08-20:
@dariolob probably a bit late I also had this issue. I just managed to get my A1016 keyboard set up in 14.6.1 with `blueutil` following [these instructions](https://forums.macrumors.com/threads/dead-apple-a1016-bluetooth-keyboard-is-there-a-way-to-revive-it.2351699/?post=31327826#post-31327826) and similar to this [comment](https://github.com/toy/blueutil/issues/71#issuecomment-1832087869) — note the _dashes_ not colon in the address, it's easy to miss.

I did, however, do a few things beforehand which may have impacted the connectivity issues:

1. Turn on the A1016 keyboard.
1. Use `--inquiry` to get the MAC address:
    ```sh
    $ sudo BLUEUTIL_ALLOW_ROOT=1 blueutil --inquiry
    address: 00-0a-95-4b-dd-b1, not connected, not favourite, not paired, name: "00:0A:95:4B:DD:B1", recent access date: 2024-08-20 02:01:23 +0000
    ```
1. Add it as a favourite:
    ```sh
    $ blueutil --add-favourite 00:0A:95:4B:DD:B1
    ```
1. Pair with 0000:
    ```
    $ blueutil --pair 00-0a-95-4b-dd-b1 0000
    Input pin 0000 on "00:0A:95:4B:DD:B1" (00-0a-95-4b-dd-b1)
    ```
    Note the command above will immediately run and not pause until you type in `0000` on the A1016.
1. Type `0000` on the top row of the A1016 (i.e., not the numeric keypad) and hit return. (You won't be typing anywhere or not into a prompt, just type it and hit enter.)
1. The "Connected" pop up appears (interestingly with a A1255 keyboard outline, not the A1016).

All seems to work now, typing this out on an A1016 😊 Hope it works for you!

toy, 2024-08-22:
@alexcu What happens if inquiry is done without sudo?

alexcu, 2024-09-16:
> @alexcu What happens if inquiry is done without sudo?

Sorry, I missed this reply, @toy. Seems to work in either case for me 🤷 

```
❯ blueutil --inquiry
2024-09-17 09:31:27.078 blueutil[74469:4271813] -[IOBluetoothDeviceInquiry initWithDelegate:] -  0x600003070040
<devices>
2024-09-17 09:31:37.616 blueutil[74469:4271813] -[IOBluetoothDeviceInquiry dealloc] -  0x600003070040
❯ sudo BLUEUTIL_ALLOW_ROOT=1 blueutil --inquiry
2024-09-17 09:32:23.768 blueutil[76254:4274982] -[IOBluetoothDeviceInquiry initWithDelegate:] -  0x600001bb4030
<devices>
2024-09-17 09:32:34.300 blueutil[76254:4274982] -[IOBluetoothDeviceInquiry dealloc] -  0x600001bb4030
```

alexcu, 2024-09-16:
> now the apple one doesn't seem so comfortable to use lol

@a2mta the exact same thing happened to me 😂 Wondering if I can take the keyboard apart and adjust some of the rubber domes.

s1eelra1, 2024-10-07:
@alexcu Thanks for the tips. I just was successfully able to pair a couple of older MS mice that are "replacement" units. I followed your instructions, and just removed the 0000 (mice can't enter that of course). This is on 14.7 as well.

trmorrisnd, 2024-12-14:
Newbie here. Not a programmer! I've got an M1 running Ventura (14.6). I want to connect a A1016 I bought at a yard-sale. 

A keyboard does show up in bluetooth-setting "nearby devices" as: jvburke's keyboard. The Connect button does nothing that I can see. 

How exactly do I follow the advice above?? E.G.: "blueutil --pair <kdb Mac>"

toy, 2024-12-15:
@trmorrisnd The behaviour you describe suggests to me either hardware/firmware problem with keyboard or some problem with the mac. Do you have access to some other mac or other bluetooth keyboard or other device to try to identify where lies the problem?

Barack02, 2024-12-17:
hello guys, I've been reading all the previous instructions to connect the wirless keyboard, I'don't undertand why it doesen't work.
So this is the resume of my process:
-I correctly installed blueutil
- I've done the permission on > setting >privacy >terminal turn on, and then see if it works by turn on and off the bluethoot whit blueutil
- I've found the ID with blueutil
-I've add the device in my favourite
-Then i tried to pair with and without the 0000 at the end
but nothing change

Some information:
- my device is a Macbook Pro with intel processor from 2018
-when i try to connect with the Bluethoot setting i see the device and i can see the pin code by tap 0000 in the keyboard but after a millisecond it disappera

toy, 2024-12-21:
@Barack02 As it doesn't work from settings too, I also suggest checking if the keyboard can be connected to another mac or if another keyboard/device can be connected to the mac

Barack02, 2025-02-21:
It actually worked!!!!! Finally!!!! thank you very much
WRITED WITH THE A1016

Orn3tte, 2025-03-04:
Wasn't working for me with my Mac mini m4 just out of the box until I installed the os update to 15.3.1.
(typed on the A1016)

sisamon, 2025-04-09:
> [@trmorrisnd](https://github.com/trmorrisnd) I have two A1016 keyboards, one is connected to my M1/Sonoma Mac, the other behaves exactly as you describe. I tried connecting it to an old MacMini running High Sierra, there I use the official process, no blueutil, then as soon as I enter the code, it connects, then the green power light goes off, and the keyboard shows as connected but does not work. I suspect something is wrong with our keyboards.

OK now the one that used to work is no longer working.

jwrw, 2025-04-19:
Upgraded Mac Studio M1 Max to Sequoia 15.4.1.  All seemed OK.  Was running fine.  Just re-booted (normally I just let it sleep but decided to give it a treat!) and keyboard is not re-connecting.  Still shows as a paired device.  Tried various on-off combinations to get it back - no luck.  Finally tried unpairing it.  It's back to the usual Mac showing a 6 digit pin which you can't enter until you press 'return' - which immediately sends the default '0000' + <return> which naturally doesn't work. Don't know if the keyboard has 'gone west' or if the upgrade is the culprit but I'm not typing this on the A1016 :-(

jwrw, 2025-04-19:
Having tried various things in various orders I tried some steps based on https://discussions.apple.com/thread/255844013?sortBy=rank.
1) Find the Mac address of the keyboard: `blueutil --inquiry 30` should give you that.
2) Delete bluetooth prefs `sudo rm /Library/Preferences/com.apple.[bB]luetooth.*`
3) Use old classic pairing `sudo defaults write /Library/Preferences/com.apple.Bluetooth.plist ClassicPairingEnabled 1` (adds one line into the prefs file.
4) Switch keyboard off then on to get into pairing mode (LED flashing)
5) Initiate pairing `blueutil --pair xx-xx-xx-xx-xx-xx 0000` (using your Mac address) - You should get prompt "Input pin 0000 on "Apple Wireless Keyboard""
6) Type 0000 <enter> (possibly a few times???)

The more sane among you will save a copy of bluetooth.prefs before deleting it and be able to put back the original settings, one by one, to see which of the new ones has broken it in the first place.  My mouse and keyboard are working OK, so I'm happy for now...

tasagore, 2025-04-21:
I've tried all those solutions but no luck :(

Sequoia 15.3.2, Macbook Pro 16" with Intel, no way to pair my A1016, I've allways get 0x1f error:

1) blueutil  --pair 00-0a-95-xx-xx-x 0000
2) I must press the enter key on the A1016 or nothing happens. When I press it it pass to step 3
3) Input pin 0000 on "00:0A:95:xx:xx:xx" (00-0a-95-xx-xx-x)
4) I press 0000+enter in the A1016
5) Failed to pair "00-0a-95-xx-xx-x" with error 0x04 (Page Timeout) or error 0x1f (Unspecified Error)

Any idea?

jwrw, 2025-12-07:
Each time I upgrade I say a little prayer to the Apple gods and so far it has worked out OK.  The A1016 is still working fine on my M1 Mac Studio under Tahoe 26.1.  One thing to note.  I've been using NiMH batteries for a while now.  These only work if they are absolutely fully charged and the keyboard won't connect if their voltage starts to drop, even a little. Try using new alkaline or zinc carbon (non-rechargeable) to eliminate that as a possible cause until you have a connection. 

N.B. My output from the command  `sudo defaults read /Library/Preferences/com.apple.Bluetooth.plist`  is...
`{
    BluetoothAutoSeekKeyboard = 1;
    BluetoothAutoSeekPointingDevice = 1;
    ClassicPairingEnabled = 1;
    move3PPLEMSToLegacyModeSerial = 1;
}
`
... if that helps at all.

clanger9, 2025-12-07:
I have a vague recollection that it won't pair if the macOS Bluetooth prefpane is open.
Have you tried it _without_ the settings window displayed?

toy, 2026-01-13:
@jolshin This is all very frustrating, as pin code is supposed to enhance security. Still, wouldn't providing empty pin `blueutil --pair <MAC address> ''` also work?