toy/blueutil #72: Discoverability issue in ventura

Rishi-DK, 2023-02-16:
I tried to set bluetooth discoverability on, but it returns failing. Is there a problem on my side? 
Can you help. I am using Macbook pro M1 with Ventura 13.2.1 
[image]

M00N-MAN, 2023-07-09:
Hello All,

So I confirm that the **Discoverability Control Feature isn't working**

I have same situation on MacBook Pro 14-inch, 2021 M1 Max, Ventura 13.3.1 (a) except below commands doesn't return any output on attempt to turn discoverability OFF. However after any subsequence of these commands macbook still discoverable on iphone/android/etc... :|

```
sudo BLUEUTIL_ALLOW_ROOT=1 blueutil -d 0
sudo BLUEUTIL_ALLOW_ROOT=1 blueutil --discoverable 0
sudo BLUEUTIL_ALLOW_ROOT=1 blueutil --discoverable 1
Failed to switch bluetooth discoverable on in 10 seconds
sudo BLUEUTIL_ALLOW_ROOT=1 blueutil --discoverable 0
sudo BLUEUTIL_ALLOW_ROOT=1 blueutil -d 1 
```

so for sure

```
sudo BLUEUTIL_ALLOW_ROOT=1 blueutil -d 0 && echo OK || echo FAIL 
OK

sudo BLUEUTIL_ALLOW_ROOT=1 blueutil -d 1 && echo OK || echo FAIL
Failed to switch bluetooth discoverable on in 10 seconds
FAIL
```

toy, 2023-07-09:
Is there a reason to use sudo? `BLUEUTIL_ALLOW_ROOT` was added specifically to make it not possible to accidentally run blueutil as root, as it is not needed and causes problems, see https://github.com/toy/blueutil/issues/41#issuecomment-727227663 (specifically https://github.com/toy/blueutil/issues/41#issuecomment-727227663 on how to cleanup).

ttzuesf, 2023-12-24:
I also has the similar issue,

```bash
Failed to switch bluetooth discoverable on in 10 seconds
```

Rishi-Sudhakar, 2024-06-15:
Can confirm, the issue is still in MacOS 15 Sequoia (Developer Beta 1)

Rishi-Sudhakar, 2024-06-15:
> Is there a reason to use sudo? `BLUEUTIL_ALLOW_ROOT` was added specifically to make it not possible to accidentally run blueutil as root, as it is not needed and causes problems, see [#41 (comment)](https://github.com/toy/blueutil/issues/41#issuecomment-727227663) (specifically [#41 (comment)](https://github.com/toy/blueutil/issues/41#issuecomment-727227663) on how to cleanup).

My guess is that, this issue is limited to Apple silicon macs. Not sure if intel users could reproduce this issue. Also apple has removed the ability to freely reset bluetooth(removing plist doesn’t really reset the bluetooth module completely), so too many issues since MacOS 13.