toy/blueutil #73: Paired with pin

ORANGEID, 2023-03-03:
Hi there

I having problems paring with pin

ie blueutil -d --pair 00-14-03-05-59-92 1234

errors with
0
Failed to pair "00-14-03-05-59-92" with error 0xe00002bc (UNKNOWN ERROR)

I can connect via the bluetooth settings with the same pin, but not with blueutil

When I try to pair using blueutil it show is Bluetooth setting "passkey mismatch", is there different format I need to do

I am running this on Mac 12.4 using blueutil 2.9.1 

Any help would be appreciated 

Cheers

James

toy, 2023-03-04:
`0xe00002bc` is a very generic `kIOReturnError`
It may be worse trying if system restart helps, or trying if there is access on another mac, though it can be one of yet unresolved with later version of OS (#62, #63, #71, #72, …)

ORANGEID, 2023-03-06:
Thanks for getting back to me, tried restart and and on few different Mac running Monterey and neither worked, ran on older Mac running Catalina and it worked first time, Bluetooth on Monterey seem to have quite a lot of bluetooth issues

toy, 2023-03-07:
> Bluetooth on Monterey seem to have quite a lot of bluetooth issues

Sadly yes