toy/blueutil #74: Connecting to a tethered Bluetooth network

shivaprsd, 2023-03-04:
Hi, thanks for this utility.

Is it possible to connect to Bluetooth PAN using `blueutil`? I mean the equivalent to right clicking on a connection and choosing 'Connect to network' in the GUI.

I think `networksetup` might be the correct tool for this; anyway thought to ask here. 🙂

toy, 2023-03-04:
Creating/enablying/connecting to device network is currently out of scope for `blueutil`. So connecting bluetooth device using `blueutil` and then probably using `networksetup` :)

closed 2023-03-04 by toy

shivaprsd, 2023-03-05:
I thought so. However since the GUI is in the Bluetooth Preferences, I thought the API could be among those `blueutil` uses.

I haven't found a way with `networksetup`. Most guides online only show how to connect to WiFi. 😕