toy/blueutil #75: Cannot pair from Monterey (12.6) to iOS (16.2) device using blueutil

navinkumarr, 2023-03-18:
### Pairing from Mac OS to iPhone is failing.

### version details
blueutil - 2.9.0
iOS - 16.2
Mac OS - Monterey 12.6

### Steps

I am able to find the bluetooth address using below command

``` sh
~ blueutil --inquiry 30 
2023-03-18 17:37:59.638 blueutil[30590:5101295] -[IOBluetoothDeviceInquiry initWithDelegate:] -  0x600002ce0030
address: 38-65-b2-cf-57-95, not connected, not favourite, not paired, name: "iPhone (65)", recent access date: 2023-03-18 12:08:12 +0000
2023-03-18 17:38:12.728 blueutil[30590:5101295] -[IOBluetoothDeviceInquiry dealloc] -  0x600002ce0030
```

On running pairing command I get pairing popup in iPhone with the same display number as prompted by `blueutil`. 
I clicked on pairing popup on the iPhone on `Pair` option and entered `yes` on terminal, but I am getting below error after that.

```
 ~ blueutil --pair 38-65-b2-cf-57-95
Does "iPhone (65)" (38-65-b2-cf-57-95) display number 809842 (yes/no)? yes
2023-03-18 17:39:45.540 blueutil[30660:5102044] -[IOBluetoothDevicePair peerPairingCompleted:withError:]: Calling connectionComplete on IOBluetoothDevicePair. Status: 31
Failed to pair "38-65-b2-cf-57-95" with error 0x1f (Unspecified Error)
```
After doing some search found one thread where they recommended to pass `0000` as the pin but that didn't work as well.

```
~ blueutil --pair 38-65-b2-cf-57-95 0000
Does "iPhone (65)" (38-65-b2-cf-57-95) display number 892102 (yes/no)? yes
2023-03-18 17:41:04.219 blueutil[30740:5102779] -[IOBluetoothDevicePair peerPairingCompleted:withError:]: Calling connectionComplete on IOBluetoothDevicePair. Status: 31
Failed to pair "38-65-b2-cf-57-95" with error 0x1f (Unspecified Error)
```

### On iPhone I get a pop up with message

> Pairing Unsuccessful
> iPhone can no longer connect to "MacBook Name".
> Forget this device and pair it again.

toy, 2023-03-18:
Thank you for reporting.
I'm wondering about the presence of log messages with bluetooth framework calls.
This may be the not yet resolved issue with later versions of OS ([see comment in previous report](https://github.com/toy/blueutil/issues/73#issuecomment-1454812953)).

mayank2498, 2023-04-19:
@toy
I tried this macOS app : https://apps.apple.com/us/app/keypad-bluetooth-keyboard/id1491684442?mt=12
It is able to establish bluetooth connection to an iOS device from Monterey 12.6.

Any pointers using which we can fix the issue in this repo, happy to create a PR for the same.

toy, 2023-04-22:
@mayank2498 For some time I have a hunch that one possible difficulty comes from blueutil being a CLI tool and something not done expectedly for bluetooth framework that is given for GUI app, but that is a hunch, so I don't have an understanding where to dig.
Another idea is about permissions, maybe worth checking which does that app get, maybe that will give an idea.