toy/blueutil #78: blueutil -p 1 returns Failed to switch bluetooth power on in 10 seconds

aleixsr, 2023-04-09:
On Ventura 13.3 i got this error : "Failed to switch bluetooth power on in 10 seconds" when running "blueutil -p 1".
I need to run it after sleep because BT doesn't wake up.

Running this script works fine, but it needs sudo:

pgrep bluetooth | xargs sudo kill
sudo launchctl list | grep -i blue | awk '{ print $3 }' | xargs sudo launchctl stop
sudo launchctl list | grep -i blue | awk '{ print $3 }' | xargs sudo launchctl start

toy, 2023-04-10:
Could you please clarify what do you mean by «BT doesn't wake up»? Was it on before sleep and is off on wake up?

toy, 2023-04-11:
Original issue feels like related to bluetooth internals, so not sure if blueutil would be able to do something. Are you able to switch power from the menu bar or preference pane after sleep? Same question for when `blueutil -p 0 && sleep 10 && blueutil -p 1` fails

aleixsr, 2023-04-12:
You're right, it is not blueutil related, I cannot turn on BT using switcher from preference pane, sorry about that.

closed 2023-04-12 by aleixsr