toy/blueutil #80: Blueutil --connect not working as a shortcut with Automator on Mac Os Sonoma beta

steveaugustine, 2023-07-28:
Blueutil --connect not working as a shortcut with Automator on Mac Os Sonoma beta

The command works normally on the terminal. It works on the automator app but it says failed to connect when it is run via a keyboard shortcut.

toy, 2023-07-28:
Can you compare permissions granted to the terminal app that you use and to Automator, and if they are different grant same to Automator?

steveaugustine, 2023-07-29:
I think both the terminal and automator have the same permissions as they seem to work fine when used independently.
The problem occurs when the automator service is run via a keyboard shortcut.
CODE

export PATH=/usr/local/bin:/opt/homebrew/bin
bluetoothStatus=$(blueutil -p)
blueutil --connect 2c-be-eb-0b-8e-61

toy, 2023-07-29:
So running shell script from Automator works. How is keyboard shortcut get assigned, built int through system preferences? As far as I know permissions of first app/binary after launchd are defining them for all children

steveaugustine, 2023-08-02:
Ya it runs through the keyboard shortcuts which can be assigned under the keyboard menu in settings. I hope the official version of mac os sonoma fixes this. Do you think u can recreate it on your pc??

toy, 2023-08-02:
Unfortunately I don't currently have access to any mac capable of running Sonoma.
I think that it may be worth trying to understand the process that is responsible for executing the automator task, simplest way that comes to mind is adding a long sleep to the shell script and checking in process tree the ancestors of it.

taybart, 2023-10-03:
I am also having this issue, since Sonoma is GA are you able to replicate it @toy ?

steveaugustine, 2023-10-03:
> I am also having this issue, since Sonoma is GA are you able to replicate it @toy ?

what version of sonoma are you on?
Right now third party apps which use blueutil like toothfairy, work inconsistently.
Real shame that apple hasn't fixed this yet.
connecting devices was so easy with blueutil.

taybart, 2023-10-03:
14.0, i'm just calling blueutil directly in a hammerspoon script

toy, 2023-10-03:
@taybart Still don't have access to a modern enough mac to be able to test it

peterfredericks, 2023-10-10:
Found a fix at least on my end. Giving permissions to Alfred 5 in the Privacy & Security → Bluetooth solved it for me.

taybart, 2023-10-11:
I already have the permissions enabled no luck :/

steveaugustine, 2023-10-11:
Same, all the perms are enabled.

steveaugustine, 2023-10-11:
It works flawlessly via the automator interface but bluetooth just turns on and off when I try to run it through a keyboard shortcut. I upgraded to the latest version of blueutil via homebrew

steveaugustine, 2023-10-11:
> Found a fix at least on my end. Giving permissions to Alfred 5 in the Privacy & Security → Bluetooth solved it for me.

Does it work as a keyboard shortcut throughj automator?

toy, 2023-10-15:
A bit of testing showed that automator quick action triggered by keyboard shortcut is executed by `com.apple.automator.runner.xpc` which is in `Automator.framework`. You can check by installing and running `pstree` in the shell of workflow (`pstree -p $$ > ~/log.txt`) or just a long sleep and finding the process in hierarchy.

Can you try giving the same permissions to that process (for me dragging to preferences the Contents/MacOS/com.apple.automator.runner binary which is inside `com.apple.automator.runner.xpc` worked, but not the xpc bundle)?